mate-desktop/mate-power-manager #129: mate-power-manager: g_variant_unref(): mate-power-statistics killed by SIGSEGV

NiceandGently, 2015-03-03:
orign report at rhbz https://bugzilla.redhat.com/show_bug.cgi?id=1190677
```
 ssabchew 2015-02-09 07:12:36 EST

Description of problem:
1. Open it 
/usr/bin/mate-power-statistics --device /org/freedesktop/UPower/devices/battery_BAT0
2. Click on LaptopBatter on the left panel
3. Click on Statistic tab on the right panel
4. Cick  on AC adapter on the left panel
5. carsh
Segmentation fault (core dumped)

Version-Release number of selected component:
mate-power-manager-1.8.1-2.fc21

Additional info:
reporter:       libreport-2.3.0
backtrace_rating: 4
cmdline:        /usr/bin/mate-power-statistics --device /org/freedesktop/UPower/devices/battery_BAT0
crash_function: g_variant_unref
executable:     /usr/bin/mate-power-statistics
kernel:         3.18.3-201.fc21.x86_64
runlevel:       N 5
type:           CCpp
uid:            1000

Truncated backtrace:
Thread no. 1 (10 frames)
 #0 g_variant_unref at gvariant-core.c:625
 #1 up_device_get_statistics_sync at up-device.c:560
 #2 gpm_stats_update_info_page_stats at gpm-statistics.c:708
 #3 gpm_stats_update_info_data_page at gpm-statistics.c:759
 #4 gpm_stats_update_info_data at gpm-statistics.c:806
 #5 gpm_stats_notebook_changed_cb at gpm-statistics.c:1121
 #15 g_object_notify_by_spec_internal at gobject.c:1149
 #16 g_object_notify at gobject.c:1197
 #17 gtk_widget_hide at gtkwidget.c:3330
 #18 gpm_stats_update_info_data at gpm-statistics.c:799
```
backtrace is at downstream link.
This is 100% reproducable with my own laptop (f21 with upower-0.99.2)
Maybe there is something missing in the port to upower-0.99.x.
Sorry, i can't check this, i don't have a linux installed with older upower version on my notebook.
Here is my own stacktrace created with gdb.
https://dl.dropboxusercontent.com/u/49862637/Mate-desktop/Bugs/mate-power-statistics

NiceandGently, 2015-03-03:
I quess this report is also related to the problem.
https://bugzilla.redhat.com/show_bug.cgi?id=1195898
But i can't reproduce it because my wireless usb mouse isn't show in mate-power-statistics.
But everytime if the statistic tab is selected and i switched back to AC adapter on the left panel, mate-power-statistics crashed.

NiceandGently, 2015-03-03:
Seems it is upower related, see https://bugzilla.redhat.com/show_bug.cgi?id=1128390#c24
I will close this report when the patch is applied.

pudh4418, 2015-03-16:
The patch is applied in upstream, please see http://cgit.freedesktop.org/upower/commit/?id=2510148b16
I have tested it, and it fix the probleam. You may close this issue now.

closed 2015-03-16 by NiceandGently