mate-desktop/mate-power-manager #135: macbook pro 6,2 backlight problem

nimalr, 2015-03-14:
mate-power-manager 1.8.0+dfsg1 on debian does not adjust the backlight properly.
There are two directories under /sys/class/backlight (gmux_backlight and nv_backlight)
Changing brightness value in nv_backlight has no effect
Adding gmux_backlight over nv_backlight in the file gpm-backlight-helper.c solves the problem for me.

```
--- a/src/gpm-backlight-helper.c        2014-03-01 17:11:07.000000000 +0530
+++ b/src/gpm-backlight-helper.c        2015-03-14 18:30:25.452728146 +0530
@@ -51,6 +51,7 @@
 
        /* available kernel interfaces in priority order */
        static const gchar *backlight_interfaces[] = {
+               "gmux_backlight",
                "nv_backlight",
                "nvidia_backlight",
                "intel_backlight",
```

I looked at gnome3 code - there the preference is selected based on the type instead of hardcoding.
Here gmux_backlight (type = platform) wins over nv_backlight (type = raw)

file plugins/power/gsd-backlight-helper.c
```
        /* search the backlight devices and prefer the types:
         * firmware -> platform -> raw */
        path = gsd_backlight_helper_get_type (devices, "firmware");
        if (path != NULL)
                goto out;
        path = gsd_backlight_helper_get_type (devices, "platform");
        if (path != NULL)
                goto out;
        path = gsd_backlight_helper_get_type (devices, "raw");
        if (path != NULL)
                goto out;
```

NiceandGently, 2015-05-20:
@nimalr 
Can you please do a pull request with adding "gmux_backlight" to gpm-backlight-helper.c ?

terinjokes, 2015-09-09:
@NiceandGently i don't think @nimalr is going to, but this also effects me. Can you go ahead and merge?

commit 7a2efb36 (2015-09-09): backlight-helper: Add gmux_backlight to backlight_interfaces

Fixes #135
closed 2015-09-09 by infirit