mate-desktop/mate-power-manager #139: mate-power-statistics produced a lot of warnings in a terminal

NiceandGently, 2015-03-27:
mate-power-statistic produce some warnings in a terminal after latest fixes for upower on systems without a battery, but no crash.
```
[rave@qemu-f22-gtk3 ~]$ mate-power-statistics 

** (mate-power-statistics:2316): WARNING **: Couldn't register with accessibility bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

(mate-power-statistics:2316): GLib-GObject-WARNING **: The property GtkButton:use-stock is deprecated and shouldn't be used anymore. It will be removed in a future version.

(mate-power-statistics:2316): GLib-GObject-WARNING **: The property GtkSettings:gtk-button-images is deprecated and shouldn't be used anymore. It will be removed in a future version.

(mate-power-statistics:2316): libupower-glib-CRITICAL **: up_device_set_object_path_sync: assertion 'object_path != NULL' failed

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'has-history' (10) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'has-statistics' (11) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'kind' (12) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'state' (13) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'percentage' (25) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'online' (7) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'update-time' (1) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'power-supply' (6) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'is-rechargeable' (9) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'is-present' (8) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'time-to-full' (24) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'time-to-empty' (23) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'technology' (14) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'capacity' (15) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'energy' (16) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'energy-empty' (17) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'energy-full' (18) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'energy-full-design' (19) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'energy-rate' (20) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'voltage' (21) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'vendor' (2) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'serial' (4) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'model' (3) was never set

(mate-power-statistics:2316): GLib-GIO-CRITICAL **: g_dbus_proxy_get_object_path: assertion 'G_IS_DBUS_PROXY (proxy)' failed

(mate-power-statistics:2316): libupower-glib-CRITICAL **: up_device_set_object_path_sync: assertion 'object_path != NULL' failed

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'has-history' (10) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'has-statistics' (11) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'kind' (12) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'state' (13) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'percentage' (25) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'online' (7) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'update-time' (1) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'power-supply' (6) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'is-rechargeable' (9) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'is-present' (8) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'time-to-full' (24) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'time-to-empty' (23) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'technology' (14) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'capacity' (15) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'energy' (16) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'energy-empty' (17) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'energy-full' (18) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'energy-full-design' (19) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'energy-rate' (20) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'voltage' (21) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'vendor' (2) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'serial' (4) was never set

(mate-power-statistics:2316): libupower-glib-WARNING **: Property ID 'model' (3) was never set

(mate-power-statistics:2316): GLib-GIO-CRITICAL **: g_dbus_proxy_get_object_path: assertion 'G_IS_DBUS_PROXY (proxy)' failed
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
```

monsta, 2015-03-27:
I confirm this in Debian Jessie, on a plain desktop (so no batteries). It started after I've updated upower from 0.9.23 to 0.99.1 (I've been holding that update before).

pudh4418, 2015-12-13:
Does this issue still exist in the current version?

monsta, 2015-12-14:
Yes, it's still the same. I get it only on real hardware though. No any warnings when running it in Virtualbox.
Also in VM it shows me AC adapter while on real hardware it doesn't:

[image]

[image]

raveit65, 2015-12-14:
yes, warnings are still there.

monsta, 2017-01-07:
Ok, it should be fixed now with https://github.com/mate-desktop/mate-power-manager/commit/ae4029805054c033dcaa01fffb3bb84b0e9ff306.

closed 2017-01-07 by monsta