mate-desktop/mate-power-manager #140: system shutdown after resume

ftoledo, 2015-03-29:
i have a Ultraslim CX 271XX netbook with Debian jessie and mate.
The system can shutdown/suspend/hibernate as normal.

But when wakeup from suspend to ram, then of 1 minute, the system instant poweroff 
This behavior  do not occur when resume from hibernate.

 I try to insolate the issue, (systemd/kernel/mate-power-manager) any help apreciated

<pre>
# dmidecode 2.12
# SMBIOS entry point at 0xdac46000
SMBIOS 2.7 present.
63 structures occupying 2565 bytes.
Table at 0x000E0840.

Handle 0x0000, DMI type 4, 42 bytes
Processor Information
	Socket Designation: CPU Socket - U3E1
	Type: Central Processor
	Family: Core i3
	Manufacturer: Intel(R) Corporation
	ID: A9 06 03 00 FF FB EB BF
	Signature: Type 0, Family 6, Model 58, Stepping 9
	Flags:
		FPU (Floating-point unit on-chip)
		VME (Virtual mode extension)
		DE (Debugging extension)
		PSE (Page size extension)
		TSC (Time stamp counter)
		MSR (Model specific registers)
		PAE (Physical address extension)
		MCE (Machine check exception)
		CX8 (CMPXCHG8 instruction supported)
		APIC (On-chip APIC hardware supported)
		SEP (Fast system call)
		MTRR (Memory type range registers)
		PGE (Page global enable)
		MCA (Machine check architecture)
		CMOV (Conditional move instruction supported)
		PAT (Page attribute table)
		PSE-36 (36-bit page size extension)
		CLFSH (CLFLUSH instruction supported)
		DS (Debug store)
		ACPI (ACPI supported)
		MMX (MMX technology supported)
		FXSR (FXSAVE and FXSTOR instructions supported)
		SSE (Streaming SIMD extensions)
		SSE2 (Streaming SIMD extensions 2)
		SS (Self-snoop)
		HTT (Multi-threading)
		TM (Thermal monitor supported)
		PBE (Pending break enabled)
	Version: Intel(R) Core(TM) i3-3217U CPU @ 1.80GHz
	Voltage: 0.8 V
	External Clock: 100 MHz
	Max Speed: 1800 MHz
	Current Speed: 1800 MHz
	Status: Populated, Enabled
	Upgrade: Socket rPGA988B
	L1 Cache Handle: 0x0002
	L2 Cache Handle: 0x0003
	L3 Cache Handle: 0x0004
	Serial Number: To Be Filled By O.E.M.
	Asset Tag: To Be Filled By O.E.M.
	Part Number: To Be Filled By O.E.M.
	Core Count: 2
	Core Enabled: 2
	Thread Count: 4
	Characteristics:
		64-bit capable
[... 1017 lines cut ...]
                description: Linux swap volume
                physical id: 1
                bus info: scsi@2:0.0.0,1
                logical name: /dev/sdb1
                version: 1
                serial: 55386987-dae3-4a66-8ed6-26f4d86f9dec
                size: 7628MiB
                capacity: 7628MiB
                capabilities: primary nofs swap initialized
                configuration: filesystem=swap pagesize=4096
           *-volume:1
                description: EXT4 volume
                vendor: Linux
                physical id: 2
                bus info: scsi@2:0.0.0,2
                logical name: /dev/sdb2
                logical name: /usr
                version: 1.0
                serial: d9cb7050-0e03-4f0d-8649-a07c12e3c218
                size: 22GiB
                capacity: 22GiB
                capabilities: primary bootable journaled extended_attributes large_files huge_files dir_nlink recover extents ext4 ext2 initialized
                configuration: created=2015-01-19 19:21:52 filesystem=ext4 lastmountpoint=/usr modified=2015-03-29 19:28:38 mount.fstype=ext4 mount.options=rw,relatime,data=ordered mounted=2015-03-29 19:28:44 state=mounted

</pre>

NiceandGently, 2015-05-20:
> I try to insolate the issue, (systemd/kernel/mate-power-manager) any help apreciated

mate-power-manager is only a front-end, all is done by systemd.
But you can disable mate-power-manger in autostart for testing and start it with a script like this:
```
#!/bin/bash

mate-power-manager --verbose >> output-m-p-m.log
```
This will create a debug output in home user folder. Make it executable add it to session start.
Than you will see what really happens. I would also try to start m-p-m with the script but not adding it to autostart. In result m-p-m won't start after suspend.
If the system shutdown now after 1 min, you will know whois the culprit, and it is triggered by systemd.

ftoledo, 2015-07-08:
maybe related: 
https://bugs.launchpad.net/ubuntu/+source/systemd-shim/+bug/1455520

monsta, 2017-03-09:
I can't reproduce it. Try using ```pm-suspend``` command-line tool instead and check the results.

monsta, 2018-01-05:
Well, no reply from the reporter. We can reopen it if the problem still happens, but I'm pretty sure the problem was somewhere in the backend, not in m-p-m.

closed 2018-01-05 by monsta