mate-desktop/mate-power-manager #142: m-p-s crash when switching from battery tab "History" or "Statistics" back to AC Adapter

sunweaver, 2015-04-03:
works:

  AC Adapter [ Details ] tab
  Battery Adapter [ Details ] tab
  Battery Adapter [ HIstory ] or [ Statistics ] tab
  back to Battery Adapter [ Details ] tab
  back to AC Adapter [ Details ] tab

fails:

  AC Adapter [ Details ] tab
  Battery Adapter [ Details ] tab
  Battery Adapter [ HIstory ] or [ Statistics ] tab
  SKIP THIS: back to Battery Adapter [ Details ] tab
  back to AC Adapter [ Details ] tab

"""
[mike@minobo mate-power-manager.pkg]$ mate-power-statistics 
Segmentation fault
"""

Is it possible, that the UI tries to acces the previously used tab name / identifier on the AC Adapter?

This is in mate-power-manager 1.8.1-5 (currently preparing that upload for Debian).

Mike

pudh4418, 2015-04-05:
I think this is a duplicate of issue #129

monsta, 2015-04-11:
@sunweaver: I guess you can close it when upower 0.99.1-3.2 reaches Jessie...

sunweaver, 2015-04-11:
Closing it now... @monsta: thanks for reminding me.

closed 2015-04-11 by sunweaver