mate-desktop/mate-power-manager #147: Brightness logic error on battery

jazzrz86, 2015-06-17:
I may have found an issue with the brightness/backlight logic when on battery:

* I have set the option reduce backlight when on battery and disabled reducing backlight when idle.
* I want the backlight to be 100.0 when on AC and 50% of that when I am on battery- I set the options in the UI.
* Initially, when I boot the machine on battery it is correct, it will halve the initial AC brightness value.  However, leaving the laptop idle will cause the backlight to increase back to 100.0 after a few seconds.

I look at the code and notice that it only considers the battery reduce option when the call to gpm_backlight_brightness_evaluate_and_set (gpm-backlight.c:272) has the use_initial = TRUE. However, looking at the idle code at idle_changed_cb (gpm-backlight.c:550) it does not pass TRUE to use_initial, cause the backlight scale to be at 100% at the end of the function- verified from using --verbose to the main exec.

I believe the use_initial flag is redundant as it should ALWAYS consider the backlight reduce on battery option.

raveit65, 2015-11-05:
https://github.com/mate-desktop/mate-power-manager/commit/954a80e6c2f945c6abf0a244d1a5dcb14c0ac5fb
PR is merged, but it is mentioned as possible fix, so i will leave the report a while open.
@jazzrz86 
Does the commit fixes your issue?

nachanon, 2015-11-05:
It is actually 9365044 (from pull #174) that tries to fix this issue.
Sorry for my mistake in the pull request comment.