mate-desktop/mate-power-manager #149: Modified AC backlight brightness setttings is not re-read when AC power was restored

nachanon, 2015-06-27:
I have experienced some inconsistency regarding how `mate-power-manager` handle power source switching:

If AC power brightness dconf key `org.mate.power-manager.brightness-ac` was changed while the machine was running on battery, plugging-in AC power won't cause the value to be re-read; it remembers the last value you last set while you were on AC power. That is, unless you changed this dconf key again after you plugged-in AC power.

To reproduce:
- Start your laptop using AC power.
- Change AC power brightness to 0% in `mate-power-preferences`; you'll see that the screen darkens.
- Unplug AC power.
- Change AC power brightness to 100% in `mate-power-preferences`; screen brightness won't change yet<sup>1</sup>.
- Plug in AC power again, you'd see that brightness *stayed the same*, instead of going back up to 100%.

MATE: MATE Power Manager 1.8.0, MATE 1.8.1 (from Debian Wheezy backports repository, with a7d224c patch)
System: Debian GNU/Linux 7.0 "Wheezy" i386
Machine: Samsung NP370R4V-S01TH Laptop

<sup>1</sup> This behavior is likely another bug (battery brightness is supposed to tie to AC brightness settings).

nachanon, 2015-09-12:
For the behavior mentioned in original post's footnote, it's now reported as bug #166.

nachanon, 2015-11-04:
See a proposed patch from bug #166: [1006_acknowledge-ac-brightness-gsettings-event-on-dc-power.patch](https://github.com/mate-desktop/mate-power-manager/issues/166#issuecomment-153547263) which also fixes this issue.

After applied this patch, plugging-in the AC power cord will now set the brightness to current value of the brightness slider in `mate-power-preferences`, rather than the value set on last time AC power was used.

MATE: MATE Power Manager 1.8.0, MATE 1.8.1 (patched against source package from Debian Wheezy backports repository, on top of a7d224c, pull #165, and #150 (comment) patch)
System: Debian GNU/Linux 7.0 "Wheezy" i386
Machine: Samsung NP370R4V-S01TH Laptop

raveit65, 2015-11-05:
https://github.com/mate-desktop/mate-power-manager/commit/954a80e6c2f945c6abf0a244d1a5dcb14c0ac5fb

closed 2015-11-05 by raveit65