mate-desktop/mate-power-manager #150: AC backlight settings was applied instead of battery, if laptop was already powered by battery from the start

nachanon, 2015-06-27:
It seems that MATE Power Manager have an inconsistency of handling battery power source at startup.

When one enabled "Reduce backlight brightness" on Battery Power tab of `mate-power-preferences`, shutdown the machine and restart it using battery power, the initial backlight brightness set by MATE Power Manager would be _the one configured in "On AC power" tab_ rather than 50% of that value; until AC power cord was plugged in and pulled out again.

To reproduce:

- Boot your laptop using AC power.
- Enable _"Reduce backlight brightness"_ in battery power tab of `mate-power-preferences`.
- Disable _"Dim display when idle"_ in battery power **and** AC power tab of `mate-power-preferences`.
- Set display brightness in AC power tab of `mate-power-preferences` to 100%.
- Shutdown the laptop.
- Remove AC power cord.
- Boot your laptop using battery power.
- Once logged in, you'd see (at a glance) that screen's backlight brightness is at the max<sup>1</sup>, rather than 50%.

MATE: MATE Power Manager 1.8.0, MATE 1.8.1 (from Debian Wheezy backports repository, with a7d224c patch)
System: Debian GNU/Linux 7.0 "Wheezy" i386
Machine: Samsung NP370R4V-S01TH Laptop

<sup>1</sup> More like it being ~50% for a brief moment at start, then gone back up to 100%.

nachanon, 2015-06-27:
Debian bug report on this issue:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=766806

nachanon, 2015-10-27:
For a reference, following is a log from `mate-power-manager --verbose` run at system's startup on battery power (and killed it ~10sec after full desktop started, to stop it from logging unnecessary output):
https://gist.github.com/nachanon/fae15ccfff672657750a

The power manager configuration at the time was 100% brightness on AC, reduce display brightness on battery, and do not dim on idle (both AC and battery).

According to the log:

* First run of `gpm_backlight_brightness_evaluate_and_set()` from `src/gpm-backlight.c` ([line 50](https://gist.github.com/nachanon/fae15ccfff672657750a#file-mate-power-manager-ref-log-L50)) seems to do things correctly.
* Second run ([line 248](https://gist.github.com/nachanon/fae15ccfff672657750a#file-mate-power-manager-ref-log-L248)) and third run ([line 307](https://gist.github.com/nachanon/fae15ccfff672657750a#file-mate-power-manager-ref-log-L307)) of it omitted battery power consideration (`2. battery scale` messages are missing).

From the few lines of logs before each run, it seems that second run is triggered as machine went into idle state, and third run is triggered as when coming back from idle state. This explains the behavior mentioned in the footnote of original post, and may indicate that this issue is closely related to bug #147 (battery brightness `use_initial` issue).

MATE: MATE Power Manager 1.8.0, MATE 1.8.1 (from Debian Wheezy backports repository, with a7d224c and pull #165 patch)
System: Debian GNU/Linux 7.0 "Wheezy" i386
Machine: Samsung NP370R4V-S01TH Laptop

nachanon, 2015-11-03:
I have checked around the 1.8.0 codebase, and found that this seems to be caused by this sequence of events:

* On system boot, `gpm_backlight_init()` in `src/gpm-backlight.c` sets `backlight->priv->master_percentage` to AC brightness set from `mate-power-preferences`.
* When machine goes into/back from idle state, `idle_changed_cb()` in `src/gpm-backlight.c` calls `gpm_backlight_brightness_evaluate_and_set()` with `use_initial=FALSE` argument.
* When `gpm_backlight_brightness_evaluate_and_set()` in `src/gpm-backlight.c` got called with `use_initial=FALSE` argument, it draws value from main brightness (`backlight->priv->master_percentage`) and applies it without considering battery dimming factor.

End result is after the system booted, and idle timer fired, backlight will be set to `backlight->priv->master_percentage` times 1.0 idle factor (due to "_Dim display when idle_" was disabled), which was essentially _the configured AC brightness_.

Solution for this would be either make idle event handler always evaluate brightness from configuration settings (which includes battery dimming). Or alternatively, apply the battery dimming to `backlight->priv->master_percentage` on boot (but side effects are to be checked).

Note: In 1.8.0 codebase, the idle event handler is _the only place_ that evaluate new brightness solely from current brightness value.

MATE: MATE Power Manager 1.8.0, MATE 1.8.1 (from Debian Wheezy backports repository, with a7d224c and pull #165 patch)
System: Debian GNU/Linux 7.0 "Wheezy" i386
Machine: Samsung NP370R4V-S01TH Laptop

nachanon, 2015-11-03:
Patch included:
[1005_make-idle-handler-always-use-configured-brightness.patch](https://gist.github.com/nachanon/da35ebff56868602020c)

This patch make the idle event handler (`idle_changed_cb()` in `src/gpm-backlight.c`) calls `gpm_backlight_brightness_evaluate_and_set()` with `use_initial=TRUE` parameter, which cause it to always consider battery power.

After applied this patch, you should see that when booted laptop on battery with "_Dim display when idle_" of Battery power unchecked, display backlight would be at 50% of configured AC-power backlight.

Also, if "_Dim display when idle_" of Battery power is **checked**, backlight would not slip back to the AC-power backlight value again; solving bug #147.

MATE: MATE Power Manager 1.8.0, MATE 1.8.1 (patched against source package from Debian Wheezy backports repository, on top of a7d224c and pull #165 patch)
System: Debian GNU/Linux 7.0 "Wheezy" i386
Machine: Samsung NP370R4V-S01TH Laptop

flexiondotorg, 2015-11-04:
Thank you for investigating this issue :-) Please can you also supply your patch as a pull request for the master branch.

nachanon, 2015-11-05:
`1005_make-idle-handler-always-use-configured-brightness.patch` is now submitted as pull #174.

Please note that although the patch applies on current master ("v1.12.0" e310320 with 11-line offset), it was tested only on MATE Power Manager 1.8.0. I cannot test on current version due to the [raised library requirement](https://github.com/mate-desktop/mate-power-manager/commit/365d495c81d21dce4ed8e494f9bb0dc2476d4535).

raveit65, 2015-11-05:
https://github.com/mate-desktop/mate-power-manager/commit/936504456393b861e26f2d69612208f364127136

closed 2015-11-05 by raveit65