mate-desktop/mate-power-manager #166: While running on battery, changes in configured AC backlight level are ignored even when battery's configuration is tied to it

nachanon, 2015-09-12:
I have encountered some strange behavior regarding how MATE Power Manager deals with display brightness when the machine is running on battery.

When I unchecked _"Reduce display brightness"_ in _"On Battery Power"_ tab of `mate-power-preferences`, display brightness is supposed to be tied directly to the configured brightness for AC power (namely, the brightness slider in _"On AC Power"_ tab). But when I adjusted the slider, display brightness didn't change, until I plugged-in the AC power and dragged the slider again.

How to reproduce:

* Plug AC power to the laptop.
* Start `mate-power-preferences` (or go to MATE > System > Preferences > Power Management)
* Go to _"On AC Power_" tab, and drag _"Set display brightness to:"_ slider to the highest value; screen backlight should change into maximum brightness.
* Go to _"On Battery Power"_ tab, uncheck _"Reduce display brightness"_ and _"Dim display when idle"_ box.
* Unplug AC power; your screen brightness shouldn't change.
* Go back to _"On AC Power"_ tab and drag _"Set display brightness to:"_ slider to the lowest value.
* You'll see that screen brightness doesn't change.

This also affects when _"Reduce display brightness"_ is checked, as the DC brightness is supposed to be a relative value (default 50%) of the configured AC brightness.

Workaroud is to plug in AC power, drag the brightness slider, then unplug again. (Plugging-in AC power alone won't work, due to bug #149) Alternative workaround is to use backlight keys for adjusting brightness instead.

MATE: MATE Power Manager 1.8.0, MATE 1.8.1 (from Debian Wheezy backports repository, with a7d224c and #165 patch)
System: Debian GNU/Linux 7.0 "Wheezy" i386
Machine: Samsung NP370R4V-S01TH Laptop

nachanon, 2015-11-04:
Patch included:
[1006_acknowledge-ac-brightness-gsettings-event-on-dc-power.patch](https://gist.github.com/nachanon/cef8e373b523bfd189e4)

This patch edited backlight GSettings event handler (`gpm_settings_key_changed_cb()` in `src/gpm-backlight.c`) to apply changes from AC backlight brightness settings regardless of what power mode it was operating on, leaving the job of calculating/deciding a correct backlight brightness for that power mode to `gpm_backlight_brightness_evaluate_and_set()`.

After applied the patch, you'll see that when the machine is _on battery power_, adjusting a backlight brightness slider in _On AC Power_ tab of `mate-power-preferences` will change the backlight brightness accordingly:

* If "_Reduce backlight brightness_" in _On Battery Power_ tab is **checked**, backlight will be half value of the slider.
* If "_Reduce backlight brightness_" in _On Battery Power_ tab is **unchecked**, brightness followed the slider exactly.

As bug #149 is also caused by lacking acknowledgement of AC brightness GSettings changes on battery power, this proposed patch will also fix that bug; plugging-in the AC power cord will now set the brightness to current value of the slider, not the value set on last time AC power was used.

MATE: MATE Power Manager 1.8.0, MATE 1.8.1 (patched against source package from Debian Wheezy backports repository, on top of a7d224c, pull #165, and https://github.com/mate-desktop/mate-power-manager/issues/150#issuecomment-153259857 patch)
System: Debian GNU/Linux 7.0 "Wheezy" i386
Machine: Samsung NP370R4V-S01TH Laptop

flexiondotorg, 2015-11-04:
Thanks for researching this issue and providing patches :-) Please can you also make a pull request against the master branch.

nachanon, 2015-11-05:
`1006_acknowledge-ac-brightness-gsettings-event-on-dc-power.patch` is now submitted as pull #175.

Please note that although the patch applies on current master ("v1.12.0" e310320), it was tested only on MATE Power Manager 1.8.0. I cannot test on current version due to the [raised library requirement](https://github.com/mate-desktop/mate-power-manager/commit/365d495c81d21dce4ed8e494f9bb0dc2476d4535).

raveit65, 2015-11-05:
https://github.com/mate-desktop/mate-power-manager/commit/954a80e6c2f945c6abf0a244d1a5dcb14c0ac5fb

closed 2015-11-05 by raveit65