mate-desktop/mate-power-manager #170: MATE Power Manager missing rights for SUSPEND/RESUME

hobbart, 2015-10-31:
Dear Folks,
I recently encountered a serious bug in the MATE Power Manager. In the case of RESUME from SUSPEND to RAM (opening the lid) it happens that the re-establishment of my WLAN connection takes more than 10 seconds, so that the router (FritzBox 6490) cannot deal with, causing a complete drop of the WLAN connection to that specific client.

An error is reported in /var/log/pm-suspend.log if the mate-power-manager is doing the RESUME from SUSPEND with unprivileged user rights.

Running hook /usr/lib/pm-utils/sleep.d/60_wpa_supplicant resume suspend:
Failed to connect to non-global ctrl_ifname: (null) error: No such file or directory
/usr/lib/pm-utils/sleep.d/60_wpa_supplicant resume suspend: success.

If the response of the mate-power-manager to opening and closing of the lid is set to "IGNORE" and an event file is put into /etc/acpi/events/lid (containing event=button/lid LID close
action=/usr/sbin/pm-suspend) and handling the RESUME process with privileged rights the error is disappearing and the WLAN connection is re-established almost immediately.

In that case /var/log/pm-suspend.log shows:
Running hook /usr/lib/pm-utils/sleep.d/60_wpa_supplicant resume suspend:
Selected interface 'wlan0'
OK
/usr/lib/pm-utils/sleep.d/60_wpa_supplicant resume suspend: success.

I hope my description is sufficient to reproduce and to fix the problem. It took me more than 3 months to find the root cause. In lots of forums there is only guesswork made on that problem.

Sven

raveit65, 2015-10-31:
m-p-m-release?
distro?
systemd or SysV init?

hobbart, 2015-11-01:
Sorry raveit65, I dont understand your abbreviations.
Please tell me how to check for systemd or SysV init.
The BUG occurs with all standard Linux Mint installations (tested!) 15.0 MATE (32bit), 17.0 MATE (64bit), 17.1 MATE (64bit) and 17.2 MATE (64bit) at different PCs (Dell XPS-L502X, ASUS-R505C, ASUS-R510, Eee-PC 1000H). It is highly reproducible and seem to me a very basic problem in all MATE distributions. The only point is, this BUG has never shown up, because the specific router I use (FritzBox 6490) is kind of over-sensitive ;-) forcing me to resolve the problem.
Sven

raveit65, 2015-11-01:
LinuxMint isn't all Linux distributions ;) This is only one....
For example i use Fedora and i never saw this with my fritz boxes.
And your linux use SysV init to boot the system, my and others use systemd which is newer.
On systems with SysV init, upower is resposable for suspend/hibernate, m-p-m is only a backend which use upower.
n Fedora/debian with systemd this is handled directly by systemd.
As you see LinuxMint is not the same like the rest of linux ;)
So, posting a report without saying which distro isn't helpful.
Or do you think only Linuxmint provide MATE ?

hobbart, 2015-11-01:
hmm und nun?

raveit65, 2015-11-01:
Please use english language here, to avoid that only i understand you ;)
Be patient, we have linuxmint guys here for try to reproduce the issue.
Btw, did you have file out a report at linuxmint bug reporting system?
For checking this is not related to a more general application in Linuxmint.

hobbart, 2015-11-01:
I also filed out a BUG report at Linux Mint Bug Tracking System (#1511997)
Hope this is the right place.
see link: https://bugs.launchpad.net/linuxmint/+bug/1511997

monsta, 2015-11-02:
@clefebvre: can you please take a look at this?