mate-desktop/mate-power-manager #183: Hibernate instead of suspend

ShayanH, 2016-05-15:
I have Manjaro 15.12 with Mate 1.12.1 on Thinkpad X250

In Mate Power Manager I have set to put computer to sleep when it is inactive for 30 minutes.
But instead of sleep, my computer goes to hibernate mode and I have problem with hibernation and my computer doesn't wake up correctly from hibernate.

I want my computer goes to sleep as I configured in Mate Power Manager, not going to hibernation.

How I can fix this problem?

Thanks in advance

monsta, 2016-05-18:
Interesting - I saw that on one of my machines and thought it's some local hardware issue on my side. The distro is different (LMDE 2, based on Debian Stable). I'll check it once more...

ShayanH, 2016-05-22:
Before Mate, I had XFCE on my machine and mentioned feature works correctly.

monsta, 2016-05-28:
Indeed, it still happens on that machine... not sure why, but Hibernate and Suspend buttons in shutdown dialog seem to be switched. :confused:

monsta, 2017-03-09:
Ok, now I don't see it anymore. Can you still reproduce it? Since you have Arch-based distro, you should have the latest MATE now.

joancf, 2017-04-13:
I have a linux Mint 18 and on battery power it hibernates instead of suspend. Does not do so in AC mode, or running ubuntu or cinnamon 
I have an SDD and 16G of Ram so I'm not very happy of writing such a quantity of data? 
I use MATE Desktop Environment 1.14.1

monsta, 2017-04-14:
Maybe try Mint 18.1 with MATE 1.16?

joancf, 2017-04-16:
yes I tried, but the same problem persists :-( 

Joan

mpmoran, 2017-08-03:
I'm on Mint 18.2 with MATE 1.18.0. Issue still exists.

flexiondotorg, 2017-08-04:
@monsta Ubuntu deliberately disables Hibernate because it is known to not work reliably. Does Mint override this?

monsta, 2017-09-03:
I'm not aware of such override, but I can see hibernate as an option in mate-power-preferences. Though Mint 18.x releases are based on Ubuntu 16.04. Maybe there was no override in that Ubuntu release?

monsta, 2017-09-03:
At least setting it to suspend on pressing Power button seems to work here.

monsta, 2017-09-04:
Suspend on idle option works in my Mint as well, so the original issue seems to be fixed.
It's a desktop machine, so I can't test the behavior on battery power.

skmkenn, 2017-12-02:
From a terminal, execute `gsettings set org.mate.power-manager action-sleep-type-battery 'suspend'` which should tell the Mate Power Manager to suspend-to-RAM instead of hibernating to disk.

monsta, 2017-12-24:
@joancf @shayanh @flexiondotorg 
Can you test what's suggested above?

bwat47, 2018-02-11:
I had the same problem. The gsettings change ghost mentioned fixed it for me. 

It seems like there is a bug if that gsettings key is set to hibernate even though the dropdown in the mate power settings for what to do on idle is set to suspend??

lukefromdc, 2018-02-11:
Keep in mind, hibernation is only possible with a swap partition at least as large as installed RAM. On machines today with very large amounts of RAM this can be an issue. Also lots of installs today are set up with no swap at all.

cantalupo555, 2018-08-07:
`systemctl suspend`
As our friend said, you need SWAP free.

Currently I follow a default:
2GB RAM = 2GB Swap
3GB RAM = 3GB Swap
4GB RAM = 6GB Swap (Above 2 GB, I add twice as much memory + 2GB Swap) (4+2)
8GB RAM = 10GB Swap (8+2)
16GB RAM = 18GB Swap (16+2)

coozoo, 2024-11-09:
> From a terminal, execute `gsettings set org.mate.power-manager action-sleep-type-battery 'suspend'` which should tell the Mate Power Manager to suspend-to-RAM instead of hibernating to disk.

Damn this old bug I'm still using this settings and it is almost 8 years  since initial report

mesov, 2025-11-20:
> Keep in mind, hibernation is only possible with a swap partition at least as large as installed RAM.

This sounds like a good reason why MATE power settings should use `suspend` and not `hibernate` as default sleep state on battery.

If this is set elsewhere, the power manager GUI could let the user know that the system will hibernate if the `gsettings` value is not modified, and ideally let the user choose whether to suspend to disk (`hibernate`) or to RAM (`suspend`).

See https://github.com/mate-desktop/mate-power-manager/issues/415.