mate-desktop/mate-power-manager #185: DPMS off settings not updated on RHEL 7.2

ziptie, 2016-05-25:
Hi, I'm running RHEL 7.2, with mate 1.12.1 from the EPEL repository.
The machine is a Lenovo Thinkpad T541, and the issue is present with the LCD or using monitor(s) plugged into the dock.  The screensaver works, but the screens never power off (DPMS off state) regardless of the timeout setting.  Manually setting dmps via `xset dpms 0 0 600` works, and the screen shuts off after the specified timeout.

Running `mate-power-preferences --verbose` from the commandline shows the following error in red:
     - Failed to get session for pid 17358: The name org.freedesktop.ConsoleKit was not provided by any .service files

This doesn't cause the program to terminate, and it happily lets me adjust timeouts for ac or battery, ie `- Changing sleep-display-ac to 1800`.  Searching for similar issues led me to [issue 71](https://github.com/mate-desktop/mate-power-manager/issues/71) which was closed in 2013, but the symptoms are similar.

Please let me know if any further details are required.

ziptie, 2016-05-25:
I've attached the output of `mate-power-manager --verbose >> mpm.txt` after killing the running mate-power-manager.
[mpm.txt](https://github.com/mate-desktop/mate-power-manager/files/282913/mpm.txt)

raveit65, 2016-05-26:
Confirmed, it does not work in my rhel installation on my notebook.

raveit65, 2016-05-29:
I updated both my notebook and my normal box, and after the update the issue is gone on both boxes.
Relevant package was a kernel update.
I  tested it with screensaver-idle =60 sec and display-on-ac=70 and display-on-battery= 70 , and it works.
After that i switched back to default settings and all works well.
The only issue i found is that sometimes only the screensaver theme blank-screen starts and not the one  that was requested, but in all test cases the screen power off here after a while.
notebook: asus FX700
PC: with asus mainboard
I don't know when this new kernel was available, but here fixes it the problem.

patrickbr, 2016-05-30:
I am having the exact same problem with mate-power-manager 1.12.1, running on a Thinkpad T530 with two monitors connected to the docking station. `xset dpms 0 0 600` works. The screensaver kicks in after the specified time, but the screens never go idle. I attached the [output](https://github.com/mate-desktop/mate-power-manager/files/289399/log.txt) of `mate-power-manager --verbose` on my machine, however, I cannot see any error messages.

@raveit65, which kernel version fixed the problem for you?