mate-desktop/mate-power-manager #187: brightness does  not return to value set by the function keys on battery

narasab, 2016-05-28:
On Mate 1.12, Linux Mint 17.3, and Ubuntu MATE 16.04

1. When on battery screen brightness drop to 50%, whick is okay.
2. After that I increase brightness with function keys to say 80%, which works ok.
3. After the screen dims, it returns back to 50% again but not 80% ie: the value set by the function keys.
4. If I disable "reduce backlight brightness" on battery and set brightness with the command 
    `gsettings set org.mate.power-manager brightness-ac 80`
then after dimming it returns to the value I set. So I think it reads the brightness-ac value when to restore brightness after dimming. So function keys aren't writing the brightness value at all.

snugglej, 2016-06-30:
I have the same problem but I also have a backlight on my keyboard. When I turn off the backlight on the keyboard and the system goes to dim the screen using the mate power-manager and then un-dim it will cause the backlight on the keyboard to also come back to the powered on state value of 1 even if it was off or at the state value of 2.

narasab, 2016-07-02:
Checked out Linux Mint 18 with Mate 1.14, same behaviour there too. This is a critical feature on a laptop.

quite, 2016-10-30:
I am here for the keyboard backlight issue. I could not find any way to turn it off in the Mate control panel, but did it the hard way by `echo 0 >/proc/acpi/ibm/kbdlight` (on this Thinkpad T450s). But, it doesn't stick. Kbdlight goes back on again typically after graphical login, after re-login to locked screen, and after activating an idle-dimmed screen.

simonsaysthis, 2016-11-09:
@quite I use a Thinkpad T460s, don't you also have the issue with brightness. I am facing both, keyboard backlight and brightness. I wish somebody had a permanent fix for this so I could use Mate instead of Unity

quite, 2016-11-11:
Quite possibly I would, but I don't use any kind of feature that dims brightness (on battery mode or otherwise), and I tend not to adjust the brightness in any manual way. (I just have DPMS turn off the display when 30m or so idle).

For reference, this issue is also discussed here:

https://bugzilla.redhat.com/show_bug.cgi?id=1348156
https://bugs.launchpad.net/dell-sputnik/+bug/1510344

simonsaysthis, 2016-11-27:
I wish somebody would look at this and provide a fix. I have submitted bug requests but they are not being confirmed.

narasab, 2018-01-13:
This bug is still valid on Mate 1.18.

simonsaysthis, 2018-01-13:
Yeah. The gift that keeps on giving

mladenradev, 2018-07-15:
Mint 19 Mate - issue still there....

simonsaysthis, 2018-07-15:
I don't think this will ever be fixed, to be honest. Been like that for years now.

PR #403 (2025-09-26): Brightness fixes
Fix a simple bug in KBD backlight toggle key handling and add some functionality to make KBD and LCD backlight control behave a bit more sane.
Fixes #187 as LCD brightness changes by function keys in battery mode are saved to config now.
It does not fix the problems with keyboard backlight described there, as I found this is a completely different issue. For example on my lenovo laptop, changing keyboard backlight by Fn+Space does not generate a scancode, thus GPM can't keep track of changes made to keyboard backlight. One might come up with some trickery to take such untracke [... cut]
closed 2025-09-26 by vkareh