mate-desktop/mate-power-manager #193: Change brightness via scrolling over tray icon

Yanpas, 2016-06-10:
Add support for changing brightness via scrolling over tray icon

monsta, 2016-06-15:
Just tested - the slider moves when I use mouse scroll, both with slider shown and hidden. Doesn't it work for you?

Yanpas, 2016-06-15:
I was talking about this [image] (battery icon in the tray). It would be handy and consume less space on the panel.

monsta, 2016-06-20:
Wait... changing brightness by scrolling over battery status icon? :confused:

Yanpas, 2016-06-20:
Yes)

monsta, 2016-06-21:
Oh come on... this makes no sense :smile:

Well, I would never thought that scrolling over battery **status** icon would make any changes in the system. I'd be surprised if it did so.
We have a separate brightness applet for that, so let's keep features where they belong.

closed 2016-06-21 by monsta

Yanpas, 2016-06-21:
OK, I thought it would have been smart feature. Less space is consumed and the same set of features persist)