mate-desktop/mate-power-manager #197: Migrate mate-power-manager to UPower 0.99 + suspend/hibernate helper/policy

geki-yaba, 2016-07-22:
XFCE did that move in 4.12, so if anyone interested may have a look there. [0] shows general UPower related changes/bugfixes, [1]-[3] show their three major steps towards upower-glib, which MATE uses already) and the helper/policy.

[0] http://git.xfce.org/xfce/xfce4-power-manager/log/?qt=grep&q=upower

[1] http://git.xfce.org/xfce/xfce4-power-manager/commit/?id=ae97be6f3500eea509d61c914e22c5355e7d57de
[2] http://git.xfce.org/xfce/xfce4-power-manager/commit/?id=8cc3a47c0265bcfb4627c1f503534dd20bfbffdc
[3] http://git.xfce.org/xfce/xfce4-power-manager/commit/?id=9ec0514fd9972373c1875ff3b80054145a445d9f

geki-yaba, 2016-09-03:
And as a side-note. Please do integrate the suspend/hibernate dbus backend into mate-session. The xfce4-session package does not call for org.freedesktop.PowerManagement/Suspend or Hibernate yet, which is causing headache for the user, see:

https://forums.gentoo.org/viewtopic-p-7961614.html#7961614

geki-yaba, 2016-12-19:
Could be even better, to have the power-helper utility in its own repository/package; not to let mate-session depend on mate-power-manager.

I would have spare time this week, maybe next, to work on this. Though, before I could start, I would need some instructions on how to proceed. Like, add to power-manager package or move into own package, change prefix of functions to 'xyz_' and whatever else there is to take care of before needing to do it twice, three times or more for stupid refactoring.

monsta, 2017-02-27:
I'm a bit confused. The title says "migrate", but the commits are just about adding support for upower 0.99 while also keeping support for older versions (which is already done in mate-power-manager). "Migrate" would mean dropping support for older upower, and that's something we're not going to do due to some distros still using 0.9.23 (e.g. Slackware).

geki-yaba, 2017-02-28:
The title says: "Migrate to UPower 0.99 plus suspend/hibernate helper". Not just UPower 0.99, which drops support for suspend/hibernate. Quite uncool. The commits show more than that.

XFCE made a standalone binary /usr/bin/xfce-power-helper (or so). That helper wraps DBUS API (policy) for suspend/hibernate.

Right now, suspend/hibernate is handled independently in mate-session and mate-power-manager, which could be united, if wanted. This can be seen in the referenced commits of XFCE.

The question is, if a helper binary plus DBUS API/policy is wanted in MATE, and if so how, or not.

geki-yaba, 2018-01-02:
It surely is possible to keep 0.9.x compatibility. See XFCE commit, which drops support for old version.
https://git.xfce.org/xfce/xfce4-power-manager/commit/?id=7c5ae951b4f3cf2392b51fb9d8d9aef7274f1d04

monsta, 2018-01-16:
If that helps a bit, in 1.18 we have support for suspend/hibernate features of ConsoleKit2, see https://github.com/mate-desktop/mate-power-manager/commit/a4d30e30fdf9ae097ced2e92c9a5130126ff9b94 and https://github.com/mate-desktop/mate-session-manager/commit/9923739fe6f161a05078340593bf6a91db13d9be. Yes, this is in two packages, but it adds the feature back for systems without logind.