mate-desktop/mate-power-manager #198: Shows as being on laptop battery when finished charging

mspaulding06, 2016-07-23:
When my laptop finishes charging, I see a popup message that says it is discharging on the laptop battery even though my laptop is still plugged in.  It never switches to say that the laptop is on A/C.
[image]

raveit65, 2016-07-23:
well, i lost my crystal ball last week ;)
and sorry w/o infos like distro, mate version, package version, systemd or not, nobody can reproduce the issue or help you.
Is presumptuous to ask it after? Isn't it?

mspaulding06, 2016-07-23:
Ah, sorry forgot about that.  You won't need a crystal ball ;)

Ubuntu-MATE 16.04.1 w/ systemd of course
Latest MATE 1.15 tagged code (built all debs myself)
mate-power-manager 1.15.0

mspaulding06, 2016-07-23:
Also noticed that here the state of the battery is shown as "discharging".
[image]

mspaulding06, 2016-07-24:
So running `upower -d` is showing that the battery is discharging as well.  I don't think this is a problem with MATE after all.  UPower is just reporting the wrong information.

raveit65, 2016-07-24:
I think nowadays upower communicate directly with systemd, and m-p-m reads it from systemd, not 100% sure.
But if upower shows the wrong information than m-p-m shows the same.
It can be also a kernel (driver) problem with your battery. And unfortunately systemd is not like systemd. Every distro use another version.
I'm using fedora 24 and i can't reproduce the wrong infos with my MSI FX700 notebook.
And i did a lot of testing the last week because of a pull request.

monsta, 2017-10-08:
Closing since it's an issue in upower or elsewhere.

closed 2017-10-08 by monsta