mate-desktop/mate-power-manager #203: brightness applet widens after pressing ESC

aykevl, 2016-10-10:
Ok, this is really strange.
I have a habit of pressing `ESC` to try to close applets, which usually has no effect (I would expect them to close, but alas). Except for mate-brightness-applet, which just gets a lot wider.
After this happens, simply clicking besides the applet (anywhere) does not have any effect anymore, you actually have to click the applet itself to close it. And when you try to use the applet again, it is still wide (although centered this time, not just expanded to the right).

I have no idea what this behavior means, but it's clearly not intended.

Additionally, there is this strange popup which says it can't connect to `mate-power-manager`, although the applet works just fine otherwise.

Here is a screenshot:

[image]

OS: Debian stretch/testing
version: 1.16.0-1

monsta, 2017-01-24:
Something went wrong in GTK+3 code... this isn't reproducible in GTK+2 build of 1.16.

monsta, 2017-02-02:
Looks like the problem was in the wrong arguments processed in the key handling callback. Not sure how it worked before...

monsta, 2017-02-02:
Ok, https://github.com/mate-desktop/mate-power-manager/pull/210 should fix it.

Some additional issues noticed even after the fix:
- no tooltip on the icon when popup is shown
- no action on enter/space when moving across panel via keyboard (a11y issue)
- no action on mouse scroll over icon when popup is hidden

Also I'm testing it on a desktop machine (no laptops around), and it allows me to change brightness value, and then it's shown in the tooltip, but actually I've expected some error message since there's no backlight or something like that... I get that error message in my ```~/.xsession-errors``` though, but that's not obvious to users.

raveit65, 2017-02-08:
> Some additional issues noticed even after the fix:
> - no tooltip on the icon when popup is shown

For me showing a notification is a redundant information if slider is shown, and i see the info there.
>- no action on enter/space when moving across panel via keyboard (a11y issue)

Not tested yet, but i agree it's an issue.
> - no action on mouse scroll over icon when popup is hidden

**Was** this possible with gtk2 ?
If not it is a feature reguest.

raveit65, 2017-02-08:
Indeed, slider isn't accessible via return/enter keystroke.

monsta, 2017-02-09:
In fact, tooltips aren't shown for every applet when a popup menu is active, not only with this applet, but with volume control as well. I think I read something about it somewhere... it's a limitation of either GTK+ or X11. So let's leave this alone for now. :slightly_smiling_face:

monsta, 2017-02-09:
Changing brightness via mouse scroll over icon works with GTK+2 when no popup is shown. Well, I can't test actual brightness change, but the slider does move, I can see that when I open the popup later. It's similar to volume control applet behavior.

It's not a new problem that it doesn't work in GTK+3. We had similar issues before, and it was usually caused by the lack of ```GDK_SCROLL_SMOOTH``` handling.

monsta, 2017-02-09:
As for the keyboard issue, here's a notification for @cwendling just in case :slightly_smiling_face:

raveit65, 2017-02-09:
> Changing brightness via mouse scroll over icon works with GTK+2 when no popup is shown. Well, I can't test actual brightness change, but the slider does move, I can see that when I open the popup later. It's similar to volume control applet behavior.

I tested this only on my PC, where brightness don't work.
Not tested on my notebook yet.

> In fact, tooltips aren't shown for every applet when a popup menu is active, not only with this applet, but with volume control as well. I think I read something about it somewhere... it's a limitation of either GTK+ or X11. So let's leave this alone for now. :slightly_smiling_face: 

Only clock applet behaviour is different.

closed 2017-02-10 by monsta

monsta, 2017-02-10:
Ok, this one is done. Maybe I'll try handling the scrolling issue later.

monsta, 2017-02-10:
Hmm, made it work with https://github.com/mate-desktop/mate-power-manager/commit/137ca1087f80c2bb5ff32d8f9466237132f2ff06. Thanks to upstream for the code... :slightly_smiling_face: