mate-desktop/mate-power-manager #215: nitpicking: Do not report battery runtime at the instant of AC disconnection

MingcongBai, 2017-04-12:
At the instant of disconnecting laptop from AC, the power manager applet on the panel reports a battery notification telling the battery life left...

I guess I could understand the intention, but every time when you do that, you simply get a report of a 100% of battery left with tens/hundreds of hours left of battery runtime...

chrisnovakovic, 2018-03-28:
The notification's not very useful even when the battery's not fully-charged:

[image]
[image]

This happens because the battery's UPower `time-to-empty` property is inaccurate immediately after AC power is disconnected. I've seen UPower gradually revise the value of that property five times in five seconds after disconnecting the power (and I bet that varies across laptops and battery ages), so it's not like it's practical to implement a dirty hack in `gpm_manager_engine_discharging_cb()` that either waits a fixed amount of time or polls the property and waits for it to have a "sensible" value before sending the notification. I second the suggestion to remove it completely: even if the notification could be made more useful (which I doubt), I don't know of any other OS or desktop environment that conspicuously notifies the user of the remaining battery life when disconnecting from AC power anyway so it's not as if it'd be a terrible loss.

chrisnovakovic, 2018-03-28:
If for whatever reason it's undesirable to remove the notification from the codebase, an alternative is to disable it via GSettings (although this will also suppress notifications about UPS devices that have begun discharging):

```
gsettings set org.mate.power-manager notify-discharging false
```

MingcongBai, 2018-03-30:
What I would suggest is to leave the notification for "discharging" in, while removing the runtime estimate.