mate-desktop/mate-power-manager #219: Provide simple indicator icon like gnome-power-manager does?

orschiro, 2017-05-21:
Dear all, 

I would like to fully substitute the gnome-power-manager with its MATE equivalent but I am lacking one feature that causes me to use both of them. 

I am in need of a simple indicator icon that shows me the state of charging and the remaining battery time. 

Would it be possible for mate-power-manager to provide such a basic indicator icon?

Yours,

Robert

gnome-power-manager indicator icon:
[image]

monsta, 2017-05-26:
There is a battery status applet in mate-applets, will it do?

orschiro, 2017-06-13:
Dear @monsta, 

I think it does!

Maybe my only remark would be to make the icon a bit prettier to fit and integrate better into the MATE desktop. 

But other than that it's fine. :-)

[image]

flexiondotorg, 2017-06-13:
I've just added `indicator-power` support to Ubuntu MATE 17.10 when an appropriate panel layout is activated via MATE Tweak.

orschiro, 2017-06-14:
@flexiondotorg thank you! Which are the appropriate panel layouts?

flexiondotorg, 2017-06-14:
Cupertino, Pantheon, Mutiny, Netbook, Redmond all support the new Indicators and I'll create a new layout called something like "Ubuntu MATE Contemporary" that will support them too.

monsta, 2017-09-03:
I wonder if the icon from the first screenshot is really from gnome-power-manager. As I see in its source and in Ubuntu package, it has just gnome-power-statistics tool now. The real power management functionality had been moved to gnome-settings-daemon at some point.

orschiro, 2017-09-03:
Actually no longer relevant. I got used to using indicator-power which received a lot of improvements. Shall I close it?

flexiondotorg, 2017-09-04:
Support for `indicator-power` is available in Ubuntu MATE when switching to a panel layout other than Traditional. If other distros want to support Indicators, that is their maintainers decision.

monsta, 2017-09-04:
Ok, I'll close this one. I just wondered which app actually displayed that icon. That is, if it was a tray icon, not a real indicator 🙂

closed 2017-09-04 by monsta