mate-desktop/mate-power-manager #225: mate-power-manager refuses to do anything post-update in Ubuntu 17.10a2.

Hebgbs, 2017-07-28:
#### Expected behaviour
`mate-power-manager` should perform as normal, without crashing

#### Actual behaviour
`mate-power-manager` doesn't work ever again because of something wicked and beyond my understanding.

#### Steps to reproduce the behaviour

- Grab either of the MATE builds from here; http://iso.qa.ubuntu.com/qatracker/milestones/379/builds
- Install it
- Update using the `update-manager` tool provided therein
- Reboot

#### MATE general version
1.18

#### Package version
1.18.0-0ubuntu2

#### Linux Distribution
Ubuntu MATE

#### Link to downstream report of your Distribution
https://bugs.launchpad.net/bugs/1707300

epgfm, 2017-07-28:
Another probably related downstream report from the Solus project:
https://dev.solus-project.com/T3525

Hebgbs, 2017-07-28:
I am open to TeamViewer if anyone wants to do that. Provide me a domain to chat privately with and I'll set things up with a fresh copy of Ubuntu MATE 17.10a2.

raveit65, 2017-07-29:
I can't reproduce a crash or a missing power applet in na-tray with an updated fedora 26 + mate-power-manager 1.18.0

raveit65, 2017-07-31:
see https://github.com/mate-desktop/mate-power-manager/issues/226
There are changes with keyboard brightness signal.
If you see such a error in logs 
gpm-kbd-backlight.c:329:gpm_kbd_backlight_on_dbus_signal: code should not be reached
downgrade upower to 0.94 for the moment.

pasikarkkainen, 2017-08-10:
I'm seeing this same mate-power-manager crash on Fedora 26 Mate. I'm using Lenovo t450s laptop.

ERROR:gpm-kbd-backlight.c:329:gpm_kbd_backlight_on_dbus_signal: code should not be reached
[1]+  Aborted                 (core dumped) mate-power-manager

Aug 10 14:31:08 laptop systemd-coredump: Process 5757 (mate-power-mana) of user 1000 dumped core.#012#012Stack trace of thread 5757:#012
#0  0x00007f4681c8d66b raise (libc.so.6)#012
#1  0x00007f4681c8f470 abort (libc.so.6)#012
#2  0x00007f46825cd70d g_assertion_message (libglib-2.0.so.0)#012
#3  0x00007f46825cd79a g_assertion_message_expr (libglib-2.0.so.0)#012
#4  0x00005625307f7203 gpm_kbd_backlight_on_dbus_signal (mate-power-manager)#012
#5  0x00007f467cd70bde ffi_call_unix64 (libffi.so.6)#012
#6  0x00007f467cd7054f ffi_call (libffi.so.6)#012
#7  0x00007f468287fb09 g_cclosure_marshal_generic (libgobject-2.0.so.0)#012
#8  0x00007f468287f30d g_closure_invoke (libgobject-2.0.so.0)#012
#9  0x00007f468289198e signal_emit_unlocked_R (libgobject-2.0.so.0)#012
#10 0x00007f468289a1a5 g_signal_emit_valist (libgobject-2.0.so.0)#012
#11 0x00007f468289ab0f g_signal_emit (libgobject-2.0.so.0)#012
#12 0x00007f4682db11bc on_signal_received (libgio-2.0.so.0)#012
#13 0x00007f4682da0a64 emit_signal_instance_in_idle_cb (libgio-2.0.so.0)#012
#14 0x00007f46825a3c27 g_idle_dispatch (libglib-2.0.so.0)#012

raveit65, 2017-08-10:
@pasikarkkainen 
 we know, downgrade to upower-0.94.x for the moment.

monsta, 2017-08-16:
Ok, 7facec455d825d317caffb8cafc4e4fec60065d9 should be enough to fix the crash. No idea if something more is needed for keyboard backlight, I don't have the hardware to test it. Leaving this open at least until someone confirms the fix.

raveit65, 2017-08-16:
@monsta 
see my comment https://github.com/mate-desktop/mate-power-manager/issues/226#issuecomment-319034935
Here i posted links to gnome fixes

raveit65, 2017-08-16:
again the gnome links
 .........and as always commiter adopt his changes to gnome-settings-daemon.
https://git.gnome.org/browse/gnome-settings-daemon/log/?qt=grep&q=keyboard
but not for other DEs , sigh.

raveit65, 2017-08-16:
Btw. i get those build warnings with upower-0.95
```
At top level:
gpm-kbd-backlight.c:431:1: warning: 'gpm_kbd_backlight_dbus_property_set' defined but not used [-Wunused-function]
 gpm_kbd_backlight_dbus_property_set (GDBusConnection *connection,
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gpm-kbd-backlight.c:408:1: warning: 'gpm_kbd_backlight_dbus_property_get' defined but not used [-Wunused-function]
 gpm_kbd_backlight_dbus_property_get (GDBusConnection *connection,
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gpm-kbd-backlight.c:355:1: warning: 'gpm_kbd_backlight_dbus_method_call' defined but not used [-Wunused-function]
 gpm_kbd_backlight_dbus_method_call (GDBusConnection *connection,
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

raveit65, 2017-08-16:
@pasikarkkainen 
Can you please test the scratch build for f26 with upower-0.95 before i push that to repos?
https://koji.fedoraproject.org/koji/taskinfo?taskID=21264244

raveit65, 2017-08-16:
@monsta 
first feedback at rhbz :-)
https://bugzilla.redhat.com/show_bug.cgi?id=1476622#c26
Thanks for the fix.
But i wouldn't close this report as more work is needed for supporting upower-0.95. Or should i file ot a extra report for this?

monsta, 2017-08-22:
Let's close this, the crash is fixed.

Not sure whether we need to handle `BrightnessChangedWithSource` signal or not.

raveit65, 2017-08-22:
I don't agree with closing as the cause of the crashes isn't fix.
mate-power-manager needs ported to upower-0.95
keyboard-backlight is broken on all popular lenovo notebooks. 
If you want to close it than please open a new report for this whith all infos from 
https://github.com/mate-desktop/mate-power-manager/issues/226#issuecomment-319034935

monsta, 2017-08-22:
If backlight is somehow broken, we need a separate report on that

lpramuk, 2017-08-22:
With kbd-backlight off or at max ```mate-power-manager``` crashes/fails to start 
With kbd-backlight at medium level  ```mate-power-manager``` runs/starts successfully  :-)
(on my T460s)

raveit65, 2017-08-22:
> If backlight is somehow broken, we need a separate report on that

Why?
Header of this report describes exactly the problem which is caused by missing support of mate-power-manager for upower-0.95.

monsta, 2017-08-25:
Well, I don't have the needed hardware anyway, so it's up to someone else to work on it.

mpeshev, 2017-08-25:
I have a similar problem running Fedora 25 on Lenovo T470p with (assumingly) the latest updates, MATE 1.18.0, UPower v.0.99.5.

I ran a manual bulk update a week ago and the battery applet was missing. I've added another widget in the meantime which is when I've discovered two separate issues: 

- There was a hiccup with some of my other options (closing the lid was suspending my system while I'd disabled that)
- The brightness function keys were no longer working

If I run the manager from the shell, it fires up after 2 or 3 attempts. The lid closing is still not working properly though - crashing the applet again, that's what I see in my log:

> ERROR:gpm-kbd-backlight.c:329:gpm_kbd_backlight_on_dbus_signal: code should not be reached

Sounds like the very same bug that @lpramuk reported 3 days ago. Let me know if I can provide additional insights or dmesg notes that would help.

raveit65, 2017-08-26:
@mpeshev 
What the hell means 'latest update`?
The important thing is the version of mate-power-manager?
Did you try https://bodhi.fedoraproject.org/updates/FEDORA-2017-837ae487a1 ?

mpeshev, 2017-08-26:
@raveit65 dude, if I knew what else to report, I would have shared it, trust me :) I had no suggested packages performing `dnf update` which is what I referred to as a latest public version.

I've tried to install your update shared above and it seems to be working pretty well. My previous version prior to the update was 1.18.0.1.fc25. I've downloaded the RPM from  https://bodhi.fedoraproject.org/updates/FEDORA-2017-837ae487a1 , installed it, rebooted and all three use cases were fixed:

- Battery applet/widget showing up on start
- Brightness applet working as expected with the function keys
- Closing the lid was not crashing the applet or forcing a soft suspend

I've even tried to suspend and wake up the system again - everything looks pretty solid.

Works for me - thanks for sorting this out and providing the beta fix :+1: :smile:

lpramuk, 2017-08-28:
Works4me mate-power-manager-1.18.0-2.fc25

o-alquimista, 2017-09-24:
mate-power-manager is not starting. Never working.

Arch Linux x86_64
mate-power-manager 1.18.0-1
upower 0.99.6-1

dedsm, 2017-09-27:
same as above, it crashes if keyboard backlight is off or max.

Arch Linux x86_64
mate-power-manager 1.18.0-1
upower 0.99.6-1

o-alquimista, 2017-09-27:
Except I don't have keyboard backlight on my laptop.

doyougnu, 2017-10-04:
Same as above, on my macbook air:

Arch Linux x86_64
mate-power-manager 1.18.0-1
upower 0.99.6-1

kikislater, 2017-10-08:
Same as above also on Asus G502VM

Arch Linux x86_64
mate-power-manager 1.18.0-1
upower 0.99.6-1

monsta, 2017-10-08:
Ask your distro's MATE maintainer to apply https://github.com/mate-desktop/mate-power-manager/commit/52a9489410297f08dc7f3a9ffe3bba0a0986d72d for now. We'll release version 1.18.1 later.

kikislater, 2017-10-08:
Done here : https://bugs.archlinux.org/task/55921

kikislater, 2017-10-09:
As a temporary workaround, downgrading with downgrader tool to : 
upower 0.99.4+12+g402640b-1
mate-power-manager 1.16.2-1
Everything is fine ! Got keyboard backlight and lcd backlight working with 4.12.14-1 kernel

catalin-hritcu, 2017-10-09:
@kikislater Tried to downgrade `upower` and `mate-power-manager` but `mate-power-manager` still crashes for me when changing screen or keyboard backlighting.

kikislater, 2017-10-09:
same version ?

catalin-hritcu, 2017-10-09:
@kikislater It seems a restart was needed? At least things seem to work for me too after restarting. (yes, same versions)

kikislater, 2017-10-09:
yes restart needed otherwise need to reload upower ... but don't know how to do. may be a simple systemd service !

monsta, 2017-11-01:
Version 1.18.1 with the crash fix is released now.

geki-yaba, 2018-01-02:
To have proper upower 0.99 support you may look at issue #197.
XFCE migrated to upower 0.99, which can be used as a base.

monsta, 2018-01-16:
We have the support, it's just that 0.99.5 added a feature that wasn't handled properly by our old code.

rsindlin, 2018-12-14:
This is still an issue in Centos 7, since EPEL does not have version 1.18.1.

As I just posted in [Stack Exchange](https://unix.stackexchange.com/questions/488056/mate-power-manager-keeps-crashing/488057#488057), the simplest workaround seems to be to un-comment `X-MATE-AutoRestart=true` from `/etc/xdg/autostart/mate-power-manager.desktop` so that it recovers from the crashes.

raveit65, 2018-12-15:
@rsindlin 
Does this scratch build from fedora build-server fix your problem?
https://koji.fedoraproject.org/koji/taskinfo?taskID=31471199

rsindlin, 2018-12-17:
@raveit65 
Yes - that does appear to fix it.  I haven't been able to trigger a crash at all.  Nice!  Thanks!

raveit65, 2018-12-18:
@rsindlin 
https://bodhi.fedoraproject.org/updates/FEDORA-EPEL-2018-022f80378a
Please leave karma.