mate-desktop/mate-power-manager #226: Crash with ERROR:gpm-kbd-backlight.c:329:gpm_kbd_backlight_on_dbus_signal: code should not be reached

petep, 2017-07-30:
I have exactly the same issue as reported in Debian:
https://www.mail-archive.com/debian-bugs-dist@lists.debian.org/msg1536182.html

But using Fedora 26 (64 bit), Mate spin:
https://spins.fedoraproject.org/mate-compiz/
mate-power-manager-1.18.0-1.fc26.x86_64
Hardware: Dell Latitude E7240

bk2204, 2017-07-30:
This is caused by mate-power-manager not handling unexpected dbus messages properly.  I am the Debian reporter and this does indeed crash when upower is updated to Debian's 0.99.5-2 from 0.99.4-4+b1.

bk2204, 2017-07-30:
I expect this is the commit that caused the problem, but I haven't tested: https://cgit.freedesktop.org/upower/commit/dbus/org.freedesktop.UPower.KbdBacklight.xml?id=112814fd4fa16782d7e07363d91616d154b77d93

raveit65, 2017-07-31:
Yes, i think so , and as always commiter adopt his changes to gnome-settings-daemon.
https://git.gnome.org/browse/gnome-settings-daemon/log/?qt=grep&q=keyboard
but not for other DEs , sigh.

fedora downstream report:
https://bugzilla.redhat.com/show_bug.cgi?id=1476622

As i do not have a notebook with keyboard brightness i can't confirm that.

raveit65, 2017-07-31:
same issue at https://dev.solus-project.com/T3525

bk2204, 2017-07-31:
I have a ThinkPad X1 Carbon.  I expect this would also be reproducible on many reasonably modern ThinkPads, since they tend to have backlit keyboards.

Blasterdick, 2017-08-07:
Same with T430, rfremix 25 x64

raveit65, 2017-08-07:
@Blasterdick 
It crashes with all notebooks with keyboard backlights.
Please read post and downgrade to upower-0.94 in fedora or other distros.
No need that i see 'Same with T430, rfremix 25 x64' notification in my mailbox ;-)
Thank you for avoiding needless mail spam....

Blasterdick, 2017-08-07:
@raveit65 downgrading to `upower-0.99.4-2` also did the trick.

raveit65, 2017-08-07:
did i say something different?
... another needless mail in my mailbox :-/

Blasterdick, 2017-08-07:
> downgrade to upower-0.94 in fedora

may be a little bit confusing when you see this:

```
$ dnf --showduplicates list upower
Last metadata expiration check: 0:00:19 ago on Mon Aug  7 09:54:13 2017.
Installed Packages
upower.x86_64                                                                        0.99.4-2.fc25                                                                        @fedora
Available Packages
upower.i686                                                                          0.99.4-2.fc25                                                                        fedora 
upower.x86_64                                                                        0.99.4-2.fc25                                                                        @fedora
upower.x86_64                                                                        0.99.4-2.fc25                                                                        fedora 
upower.i686                                                                          0.99.5-1.fc25                                                                        updates
upower.x86_64                                                                        0.99.5-1.fc25                                                                        updates

```

So yeah, you said something different. And be more patient and polite, please.

raveit65, 2017-08-07:
ok, unsuscribe from this report ;-)

overconscious, 2017-08-13:
I have found workaround.

Here is the solution:
Run dconf Editor and set the following:
org.mate.power-manager kbd-backlight-battery-reduce false
org.mate.power-manager kbd-brightness-dim-by-on-battery 0
org.mate.power-manager kbd-brightness-dim-by-on-idle 0
org.mate.power-manager kbd-brightness-on-ac 0

Also run mate-power-manager and uncheck:
"Dim display when idle" (if checked, it will cause mate-power-manager to crash when idle).

I have Dell L421x with  keyboard backlight and I had the same issue. After changing the above settings, mate-power-manager runs and works normally.
I have Debian Sid with  upower 0.99.5-3 and mate-power-manager 1.18.0-1.

monsta, 2017-08-16:
Same backtrace as in https://github.com/mate-desktop/mate-power-manager/issues/225 and its downstream Ubuntu report, so let's move there.

closed 2017-08-16 by monsta

lpramuk, 2017-08-22:
With kbd-backlight off or at max ```mate-power-manager``` crashes/fails to start 
With kbd-backlight at medium level  ```mate-power-manager``` runs/starts successfully
(on my T460s)