mate-desktop/mate-power-manager #228: Cannot change brightness through Power Manager or Brightness applet

yzhxxzxy, 2017-08-14:
I just upgraded Fedora from 25 to 26, with mate updated from 1.16 to 1.18. Then I cannot change any more the screen brightness through neither the Power Manager nor the Brightness applet.

raveit65, 2017-08-14:
Why not using the template to file out a report?
Try again...

closed 2017-08-14 by raveit65

yzhxxzxy, 2017-08-14:
Sorry for the informal expression.
I found that running mate-power-manager by hand can solve this problem.

yzhxxzxy, 2017-08-14:
Sorry for the informal expression.
I found that mate-power-manager does not run when I log in, so running mate-power-manager afterwards can solve this problem.

monsta, 2017-08-21:
If it's not running, you've probably hit https://github.com/mate-desktop/mate-power-manager/issues/225.

yzhxxzxy, 2017-08-22:
Thank you @monsta 
I also found that it is just the same problem as in #226