mate-desktop/mate-power-manager #233: Compiling mate-power-manager 1.18.1

Justinzobel, 2017-11-02:
Was updating the package for Solus and came across the error:

```
make -j1
make  all-recursive
make[1]: Entering directory '/home/build/YPKG/root/mate-power-manager/build/mate-power-manager-1.18.1'
Making all in src
make[2]: Entering directory '/home/build/YPKG/root/mate-power-manager/build/mate-power-manager-1.18.1/src'
libtool --mode=execute dbus-binding-tool	\
	--prefix=gpm_manager			\
	--mode=glib-server			\
	--output=org.mate.PowerManager.h	\
	./org.mate.PowerManager.xml
WARNING: Using --header and --body at the same time time is deprecated; use --body --prototypes instead
INFO: Reading /tmp/dbus-binding-tool-c-marshallers.0UMW8Y...
libtool --mode=execute dbus-binding-tool	\
	--prefix=gpm_backlight			\
	--mode=glib-server			\
	--output=org.mate.PowerManager.Backlight.h	\
	./org.mate.PowerManager.Backlight.xml
WARNING: Using --header and --body at the same time time is deprecated; use --body --prototypes instead
INFO: Reading /tmp/dbus-binding-tool-c-marshallers.GN4X8Y...
libtool --mode=execute dbus-binding-tool	\
	--prefix=gpm_kbd_backlight			\
	--mode=glib-server			\
	--output=org.mate.PowerManager.KbdBacklight.h	\
	./org.mate.PowerManager.KbdBacklight.xml
WARNING: Using --header and --body at the same time time is deprecated; use --body --prototypes instead
INFO: Reading /tmp/dbus-binding-tool-c-marshallers.VKNT8Y...
make  all-am
make[3]: Entering directory '/home/build/YPKG/root/mate-power-manager/build/mate-power-manager-1.18.1/src'
  CC       egg-color.o
  CC       egg-debug.o
  CC       egg-precision.o
  CC       egg-array-float.o
  CC       egg-idletime.o
  CC       egg-discrete.o
  CC       egg-console-kit.o
  CC       gpm-common.o
gpm-common.c: In function 'gpm_help_display':
gpm-common.c:148:2: warning: 'gtk_show_uri' is deprecated: Use 'gtk_show_uri_on_window' instead [-Wdeprecated-declarations]
  gtk_show_uri (NULL, uri, GDK_CURRENT_TIME, &error);
  ^~~~~~~~~~~~
In file included from /usr/include/gtk-3.0/gtk/gtk.h:196:0,
                 from gpm-common.c:28:
/usr/include/gtk-3.0/gtk/gtkshow.h:33:10: note: declared here
 gboolean gtk_show_uri  (GdkScreen   *screen,
          ^~~~~~~~~~~~
  CC       gpm-brightness.o
gpm-brightness.c: In function 'gpm_brightness_update_cache':
gpm-brightness.c:842:3: warning: 'gdk_display_get_screen' is deprecated [-Wdeprecated-declarations]
   gscreen = gdk_display_get_screen (display, screen);
   ^~~~~~~
In file included from /usr/include/gtk-3.0/gdk/gdkscreen.h:32:0,
                 from /usr/include/gtk-3.0/gdk/gdkapplaunchcontext.h:31,
                 from /usr/include/gtk-3.0/gdk/gdk.h:32,
                 from /usr/include/gtk-3.0/gdk/gdkx.h:28,
                 from gpm-brightness.c:34:
/usr/include/gtk-3.0/gdk/gdkdisplay.h:56:13: note: declared here
 GdkScreen * gdk_display_get_screen         (GdkDisplay  *display,
             ^~~~~~~~~~~~~~~~~~~~~~
  CC       gpm-marshal.o
  CC       gpm-upower.o
gpm-upower.c: In function 'gpm_upower_get_device_summary':
gpm-upower.c:222:51: warning: comparison of constant '2' with boolean expression is always false [-Wbool-compare]
   if (state == UP_DEVICE_STATE_CHARGING || !state == UP_DEVICE_STATE_DISCHARGING) {
                                                   ^~
gpm-upower.c:222:51: warning: logical not is only applied to the left hand side of comparison [-Wlogical-not-parentheses]
  AR       libgpmshared.a
ar: `u' modifier ignored since `D' is the default (see `U')
[... 126 lines cut ...]
gpm-kbd-backlight.c:380:41: note: each undeclared identifier is reported only once for each function it appears in
    g_debug ("signal '%s' not handled!", signal_name);
                                         ^
/usr/include/glib-2.0/glib/gmessages.h:349:32: note: in definition of macro 'g_debug'
                                __VA_ARGS__)
                                ^~~~~~~~~~~
At top level:
gpm-kbd-backlight.c:419:1: warning: 'gpm_kbd_backlight_dbus_property_set' defined but not used [-Wunused-function]
 gpm_kbd_backlight_dbus_property_set (GDBusConnection *connection,
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gpm-kbd-backlight.c:396:1: warning: 'gpm_kbd_backlight_dbus_property_get' defined but not used [-Wunused-function]
 gpm_kbd_backlight_dbus_property_get (GDBusConnection *connection,
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gpm-kbd-backlight.c:343:1: warning: 'gpm_kbd_backlight_dbus_method_call' defined but not used [-Wunused-function]
 gpm_kbd_backlight_dbus_method_call (GDBusConnection *connection,
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[3]: *** [Makefile:1299: mate_power_manager-gpm-kbd-backlight.o] Error 1
make[3]: Leaving directory '/home/build/YPKG/root/mate-power-manager/build/mate-power-manager-1.18.1/src'
make[2]: *** [Makefile:932: all] Error 2
make[2]: Leaving directory '/home/build/YPKG/root/mate-power-manager/build/mate-power-manager-1.18.1/src'
make[1]: *** [Makefile:471: all-recursive] Error 1
make[1]: Leaving directory '/home/build/YPKG/root/mate-power-manager/build/mate-power-manager-1.18.1'
make: *** [Makefile:402: all] Error 2

```

monsta, 2017-11-02:
```C
gpm-kbd-backlight.c: In function 'gpm_kbd_backlight_dbus_method_call':
gpm-kbd-backlight.c:380:41: error: 'signal_name' undeclared (first use in this function)
    g_debug ("signal '%s' not handled!", signal_name);
```

Sounds like https://github.com/mate-desktop/mate-power-manager/commit/52a9489410297f08dc7f3a9ffe3bba0a0986d72d was placed into wrong function.
It should be in `gpm_kbd_backlight_on_dbus_signal`. Check your sources.

Justinzobel, 2017-11-02:
Got it directly from https://pub.mate-desktop.org/releases/1.18/mate-power-manager-1.18.1.tar.xz

monsta, 2017-11-02:
The code is alright there (otherwise I'd got build errors while making the tarball).

Justinzobel, 2017-11-03:
Sorry my bad, I forgot to remove a conflicting patch.

closed 2017-11-03 by Justinzobel

raveit65, 2017-11-03:
> Sorry my bad, I forgot to remove a conflicting patch.

Is there a chance to check all possible local user errors before posting here a bug report next time?
Some ppl from MATE crew are getting email notification from every issue.........
.......i don't want to miss a serious bug from a user.
Also, keep in mind that github here is for bug reporting and not a user help line.
Sorry, we don't have so much man power, please use distro forums for that.
For the record, we do a QA testing before we release tarballs, and some of us are building the whole MATE stack for fedora and ubuntu since several years.
So, it isn't very realistic that you'll find an unknown and not fixed compiling issue.
Thank you.

Justinzobel, 2017-11-04:
@raveit65 I maintain the MATE desktop for Solus, sorry for wasting your time. I also manage bug reports on the Solus bug tracker which gets a lot of bug reports too. Apologies for not being perfect.

raveit65, 2017-11-04:
> I maintain the MATE desktop for Solus

Good too know, you're welcome :-)

Justinzobel, 2017-11-04:
Neither, independent.

raveit65, 2017-11-05:
So, i's a secret which packamanager solus used and you compiled always by hand :-)

Justinzobel, 2017-11-05:
Solus uses eopkg and our packing process is fairly automated.