mate-desktop/mate-power-manager #237: Cannot set the screen brightness

xoma0, 2017-12-19:
Hello,
I cannot set the screen brightness in mate power manager. However it can be set  in sysfs.
`$ cat /sys/class/backlight/acpi_video0/brightness `
`2`
`$ cat /sys/class/backlight/acpi_video0/max_brightness `
`9`
`$ ls /sys/class/backlight/acpi_video0`
`actual_brightness  brightness  max_brightness  subsystem  uevent`
`bl_power           device      power           type`

#### Expected behaviour
Screen brightness is changed when I move 'Set display brightness to:' slider.

#### Actual behaviour
Nothing happens.

#### MATE general version
MATE 1.18.0

#### Package version
mate-power-manager 1.18.1-1+sonya

#### Linux Distribution
Linux Mint 18.3 Sylvia 32-bit

Laptop: Acer Aspire 6935G
Linux kernel: 4.10.0-42-generic i686
NVIDIA GeForce 9600M GT
Driver nvidia 340.102 (proprietary)

I had Linux Mint 17 on this machine and brightness control worked properly then.

lenhart, 2018-01-12:
Hello, I had the same behavior using the proprietary nvidia driver. 
The nouvea driver (after a reboot) is able to switch the brightness.
In both cases setting the brightness before logging in works fine, but afterwards problems start.
If you change to nouveau and experience the brightness flashing to its max value in between every step that we got the same issue.

xoma0, 2018-01-13:
Now I'm using free driver and I still have this issue. Brightness control doesn't work at all.

> So there are maybe multiple parties blocking each other?
What parties do you mean? I'm new to it, could you explain it more detailed?

Mate-power-manager uses "brightness" file to set screen brightness, doesn't it?

lenhart, 2018-01-13:
Hey xoma0, 
I was half addressing some developers as well (hopefully providing some insights).
Yes the mate-power-manager uses the brightness file, but I suspect (I am not involved either, but also new with similar problems, so I thought I'd reply) that there is some other function/program/service which is responsible for setting the screen brightness from there. 

When I executed `$ cat /sys/class/backlight/acpi_video0/brightness`, changed the brightness with the slider and then executed the cat command again, I got a different value although the visible screen brightness stayed the same.  So I figured the problem lay somewhere deeper.. 
But I am only guessing as well. There are plenty of posts about this issue, I had to try a bit around until I found something which helped. 
Can you change brightness before you log onto your pc?

xoma0, 2018-01-14:
>that there is some other function/program/service which is responsible for setting the screen brightness from there.

And it locks the file? 

When I move the slider, brightness file value stays the same. However when I edit the file manualy, actual brightness is changed.

xoma0, 2018-01-14:
> Can you change brightness before you log into your account?

I don't know how to do that. I have to log in to do anything on my computer.

lenhart, 2018-01-14:
On this thread on [askubuntu](https://askubuntu.com/questions/762764/cant-change-brightness-in-ubuntu-16-04-lts) there is suggested to use an alternative program to set the screen brightness. Maybe that works for you. Apart from that, I cannot really suggest anything further. With the clarification your issue does not seem to be related to mine. Good luck though!

xoma0, 2018-02-25:
I've downgraded the kernel to version **4.10.0-38-generic**, and now I also have `nv_backlight` directory in `/sys/class/backlight/` along with `acpi_video0`. And when I move the slider in Power Manager number in `/sys/class/backlight/nv_backlight/brightness` file is changed according to the slider position. However actual brightness is still regulated in `/sys/class/backlight/acpi_video0/brightness` file. So I suppose it tries to edit wrong file. Does anyone know how to explain MATE Power Manager, which file it should use to set screen brightness? Maybe I have to edit some configuration files?