mate-desktop/mate-power-manager #238: inhibit-applet: background not drawn properly

muesli4, 2018-01-13:
#### Expected behaviour
Only one of two icons is visible.

#### Actual behaviour
Background is not drawn properly, the icon of the other state shines through the alpha channel of the current icon.

#### Steps to reproduce the behaviour
Use icons that are visible through each other (alpha channel) and set panel background to **none**.

#### MATE general version
Latest version from git.

#### Other details
I identified the bug location:
https://github.com/mate-desktop/mate-power-manager/blob/master/applets/inhibit/inhibit-applet.c#L280

That case (`PANEL_NO_BACKGROUND`) is simply not handled. I tried filling with the color but that doesn't seem to solve it.

lukefromdc, 2018-01-13:
Confirmed here with mate-panel 1.19.4 from git master on Debian Unstable

dagobertstyle, 2018-01-22:
Confirmed on LinuxMint18.3, mate-panel 1.18.6, inhibit-applet 1.18.1

lukefromdc, 2018-01-22:
I can confirm the issue exists only with the system background and not image nor color custom backgrounds, consistant with what the report that this case is not handled in drawing the background would be expected to do.

lukefromdc, 2018-01-22:
It looks like with the default background handling (nothing in the applet code being applied), what is being picked up when the background is redrawn is the background plus the icon, thus the icon can be rerendered but not erased. Hardocding a transparent background does NOT work, I just tried it. The code draws the background and the icon to the same window, meaning that the default case seems to be using what was last on that window icon and all) as the source for the next rendering of the background. Thus the icon is picked up and "merged" with the background.

muesli4, 2018-01-22:
I'm willing to provide a fix but I don't know how to get the correct background. I noticed that in case of `PANEL_NO_BACKGROUND` `mate_panel_applet_get_background` just zeroes out both pattern and color so they are of no use.

lukefromdc, 2018-01-23:
This works in gnome-power-manager, so I will see if I can find the commit in question. Possible it might have to become in an-process applet for this,depending on what I find

lukefromdc, 2018-01-23:
Try https://github.com/mate-desktop/mate-power-manager/pull/240 which fixes the icon problem by using a GtkImage to display it, rather than drawing it directly on the same widget carrying the background as before

closed 2018-01-24 by lukefromdc