mate-desktop/mate-power-manager #239: Switch from libgnome-keyring to libsecret

sunweaver, 2018-01-15:
The GNOME people in Debian are asking for replacement of libgnome-keyring in mate-power-manager and use libsecret instead.

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=867937

They even started raising severity which threatens mate-power-manager to be auto-removed from Debian testing. Any porting work on this in sight?

monsta, 2018-01-15:
Oops. I guess there's no patch yet?

lukefromdc, 2018-01-16:
This comes right before feature freeze for 1.20. What kind of time pressure from Debian are we talking about? Last time this happened, it was libunique->GtkApplication and took months of work.   I have zero starting point knowledge for this one.

sunweaver, 2018-01-16:
Hi

On Tuesday, January 16, 2018, lukefromdc wrote:
> This comes right before feature freeze for 1.20. What kind of time pressure from Debian are we talking about? Last time this happened, it was libunique->GtkApplication and took months of work. I have zero starting point knowledge for this one. 

We have worked around this issue in Debian by building with the --without-keyring option. This switch disable keyring locking before suspending/screen-locking, if I understood the code correctly.

The code passages are small and many other peole are currently migrating to libsecret. I bet there are many PRs out there against other software where we could learn from.

Unfortunately, the code in m-p-m that needs to be ported has been removed from gnome-power-manager before gpm was ported away from libgnome-keyring.

Conclusively, the issue is not that pressing anymore from the Debian perspective, but should be addressed sooner or later...

Mike 

-- 
Sent from my Fairphone 2 (running Sailfish OS)

monsta, 2018-01-16:
Ok, good to know it's not so urgent anymore. We'll try to port it later, but it's not clear how... We need to migrate from `gnome_keyring_lock_all_sync` function, and [this doc](https://developer.gnome.org/libsecret/unstable/migrating-locking.html) just says "no equivalent, use platform specific DBus APIs".

raveit65, 2018-01-16:
> Unfortunately, the code in m-p-m that needs to be ported has been removed from gnome-power-manager before gpm was ported away from libgnome-keyring.

Code was moved to gnome-settings-daemon and/or control-center with gnome3.
https://git.gnome.org/browse/gnome-settings-daemon/tree/plugins/power
You should find hints in their git log.
Search for 'power'.

raveit65, 2018-01-16:
Without searching for more hints it looks like that gnome-keyring is removed from power plugin.
https://git.gnome.org/browse/gnome-settings-daemon/commit/?id=08d270566709bb4b25ff6c6e79848a7f8441847f

sunweaver, 2018-01-16:
On  Di 16 Jan 2018 11:27:14 CET, monsta wrote:

> Ok, good to know it's not so urgent anymore. We'll try to port it
> later, but it's not clear how... We need to migrate from
> `gnome_keyring_lock_all_sync` function, and [this [...]

You could copy over the locking code from libgnome-keyring as is and  
ship it with m-p-m.

Or you could drop keyring locking as a feature entirely.

Mike
-- 

DAS-NETZWERKTEAM
mike gabriel, herweg 7, 24357 fleckeby
mobile: +49 (1520) 1976 148
landline: +49 (4354) 8390 139

GnuPG Fingerprint: 9BFB AEE8 6C0A A5FF BF22  0782 9AF4 6B30 2577 1B31
mail: mike.gabriel@das-netzwerkteam.de, http://das-netzwerkteam.de

monsta, 2018-01-19:
Ok, we'll try to come up with something after we release 1.20.

vaurelios, 2018-06-01:
> Or you could drop keyring locking as a feature entirely.
I agree with that.

raveit65, 2018-08-13:
> Ok, we'll try to come up with something after we release 1.20.

Any progress or an idea to do that?

sunweaver, 2018-08-13:
For Debian builds, we disable keyring locking...
https://bugs.debian.org/cgi-bin/bugreport.cgi?att=1;bug=867937;filename=no-keyring.patch;msg=24

raveit65, 2018-08-13:
@sunweaver 
Does that mean that session isn't locked when coming from suspend/hibernate?
Or what is disabled in practice?

monsta, 2018-08-13:
The `lock-keyring-suspend` key has this description:

> Whether the GNOME keyring is locked before the computer enters suspend. This means the keyring will have to be unlocked on resume.

Same goes for `lock-keyring-hibernate` key which is for similar locking before hibernation.

monsta, 2018-08-13:
I can't find where g-s-d or any other part of their software does the same things as in `gpm_control_suspend`/`gpm_control_hibernate` functions here... I know all that had been removed from g-p-m long ago, that's all.

lukefromdc, 2019-12-07:
We now have a problem: gnome-keyring still exists, but the libgnome-keyring and libgnome-keyring-dev packages needed to build and run gnome-keyring are no longer provided in Debian Unstable

vaurelios, 2020-03-25:
@lukefromdc 

> libsecret is a library for storing and retrieving passwords and other secrets.

...

> gnome-keyring and ksecretservice are both implementations of a Secret Service.

...

> libsecret replaces libgnome-keyring.

[Source](https://wiki.gnome.org/Projects/Libsecret)

So, what must be replaced is `libgnome-keyring` by `libsecret` not the `gnome-keyring` it still needed.

vaurelios, 2020-03-31:
m-p-m only uses libgnome-keyring for locking, is that true?

I know that gnome-keyring is automatically unlocked in most cases, by PAM or by DM.

Does anybody know if its automatically locked too? if everything is true, the `libgnome-keyring`-related code can be removed and there's also no need for `libsecret`.

mattst88, 2020-06-14:
Gentoo would like to remove libgnome-keyring since it's been deprecated for a long time. Would be nice to at least have this fixed upstream.

paper42, 2022-04-28:
Can this be closed? #351 was merged and the commit is in 1.26.0

rbuj, 2022-04-29:
@paper42 Thank you for the notice!

closed 2022-04-29 by rbuj