mate-desktop/mate-power-manager #243: Brightness and Inhibit action help does not open unless mate-power-manager help opened first

lukefromdc, 2018-01-31:
#### Expected behaviour
Brightness and Inhibit action help opens if help menu item clicked for the applet in question

#### Actual behaviour
Neither applet's help opens UNLESS mate-power-manager's help has been opened first

#### Steps to reproduce the behaviour
Add the brightness or inhibit applet to the panel, click on the applet, click on the "help" menu item. Try this without having first opened mate-power-manager's help. Now run "killall mate-panel" and click on mate-power-manager's tray icon. Click "preferences" in the munu, and in the dialog click the help button. Close the help window. Now repeat with either applet and help opens.

#### MATE general version
1.20 for this package and most of today's 1.20 git master packages, rest are still at 1.19 from git master

#### Package version
mate-power-manager built from 1.20.0 release git commit point

#### Linux Distribution
Debian Unstable

#### Link to downstream report of your Distribution
None is this is a locally compiled package

raveit65, 2018-01-31:
I can confirm the issue with the applets with master branch. Help doesn't open and yelp itself i crashing here and generates a abrt alarm in fedora.
Indeed, after help is called one time from m-p-m gui it opens fine from applets too.

monsta, 2018-01-31:
Not our problem guys... it's the same old and unfixed https://bugzilla.gnome.org/778553 😞

raveit65, 2018-01-31:
Lol, you reported it already :-)

monsta, 2018-01-31:
Yeah, that's my report. No replies as you can see...

monsta, 2018-02-08:
I suggest closing this, as it's a problem in Yelp.

lukefromdc, 2018-02-08:
Didn't we find some workaround for this a while back when testing something else?

closed 2018-02-08 by lukefromdc

monsta, 2018-02-20:
I only remember https://github.com/mate-desktop/mate-applets/pull/262 where a link was temporarily changed for testing. But it was only for testing.