mate-desktop/mate-power-manager #248: high cpu usage after migration to 1.20

alexarnaud, 2018-02-16:
#### Expected behaviour
Same behavior as in version 1.18

#### Actual behaviour
/usr/lib/mate-power-manager/mate-inhibit-applet uses about 6% of CPU on the machine of a Debian user.

#### Steps to reproduce the behaviour
Just update from Mate Power Manager 1.18 to 1.20

#### MATE general version
1.20

#### Package version
1.20.0-1

#### Linux Distribution
Debian Sid

#### Link to downstream report of your Distribution
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=890603

alexarnaud, 2018-02-17:
User message are

> ** (mate-power-manager:1178): WARNING **: Couldn't register with
> accessibility bus: Did not receive a reply. Possible causes include:
> the remote application did not send a reply, the message bus security
> policy blocked the reply, the reply timeout expired, or the network
> connection was broken.

@lukefromdc: Any idea? Do we check if accessibility is enabled or not?

raveit65, 2018-02-17:
If i start  the applet only from command line (not loaded with mate-panel-test-applets)
top shows me those values for the applet and X.
'top -p 31446,1015'   (pids can be differ)
[image]
after loading with mate-panel-test-applets
[image]

raveit65, 2018-02-17:
By the way, does this applet really do something nowadays?

lukefromdc, 2018-02-17:
Yes it does; It provides the brightness and inhibit applets (they work), and it also provides a battery charge monitor. The latter function overlaps with the "battery charge monitor" applet which offers more options in how to display laptop battery status. Also can set policies for dealing with low battery and similar events. A buggy BIOS in my laptop makes that last item not usable, as that laptop will throw random warnings of zero battery left that will cause a shutdown if m-p-m is set to suspend or shut down on low battery and if it is set to do nothing throw notifications. The applet is good, the laptop BIOS is bad but even the buggy BIOS is enough to prove the applet works if the underlying firmware is good

jonathonf, 2018-04-03:
Just to say this exists on both Manjaro and with custom Ubuntu 16.04 packages. Adding `mate-inhibit-applet` to the panel leads to ~5% CPU use and and extra ~1 load (on a variety of Intel i7 quad-core and i5 dual-core CPUs).

lukefromdc, 2018-04-04:
I do see the inhibit applet on an AMD 8 thread bulldozer chip sometimes at 0.5% CPU, at times where "web content" from Firefox is at 0.75%CPU. Very powerful system here, but the inhibit applet should never use 2/3ds the CPU of Firefox or any portion therof.

jonathonf, 2018-04-04:
Comparing some `strace` output from various applets (`mate-power-manager`, `mate-brightness-applet`, `mate-netspeed-applet`, and `mate-inhibit-applet`) `mate-inhibit-applet` shows a constant (high frequency) polling ([system calls](https://gist.github.com/jonathonf/c68decb7f8f434d5c46bf3ab248dff0a), [network](https://gist.github.com/jonathonf/61bf422f41e611dc87fbc411a5cf2565)). `mate-netspeed-applet` only polls and updates once per second (as expected), the other applets show activity only when interacted with.

Checking with `gdb`, the polling is happening during `g_main_loop_run()`:

```
#0  0x00007fe7b0b2397b in poll () at /usr/lib/libc.so.6
#1  0x00007fe7b1059613 in  () at /usr/lib/libglib-2.0.so.0
#2  0x00007fe7b10599d2 in g_main_loop_run () at /usr/lib/libglib-2.0.so.0
#3  0x00007fe7b1d8241f in gtk_main () at /usr/lib/libgtk-3.so.0
#4  0x00007fe7b2473152 in  () at /usr/lib/libmate-panel-applet-4.so.1
#5  0x00005608cc61d3c5 in  ()
#6  0x00007fe7b0a58f4a in __libc_start_main () at /usr/lib/libc.so.6
#7  0x00005608cc61d40a in  ()
```

Could this be solved/reduced by a `sleep()`?

raveit65, 2018-04-29:
I just tested it with my f27  installation and i couldn't reproduce it with AMD Ryzen 5 1400 and nvidia graphic.

raveit65, 2018-04-29:
Inhibition applet runs here in fedora 28 with 2-3 % cpu with my AMD 4 core 8 thread.
Seems this is equivalent to 7-9 % with an old i3 proc from @pyaazkaparatha 
I would expect that this applet should run  with lower cpu load.
So let's see what is different to m-p-m-1.18.x

pyaazkaparatha, 2018-04-29:
Does it increases the Xorgs consumption too @raveit65 ?

raveit65, 2018-04-29:
Yes, with the applet X runs with 5-6% cpu load, without the appet X have 3-4 cpu load.

raveit65, 2018-04-29:
It looks like that inhibition applet from m-p-m-1.19.1 doesn't increase the cpu load.
@pyaazkaparatha 
Can you downgrade to this version from fedora koji build-server to confirm, please?
https://koji.fedoraproject.org/koji/buildinfo?buildID=1012505
Or downgrade with **dnf**, not sure if this version is available at repos.

pyaazkaparatha, 2018-04-29:
I added back applet. The CPU usage rose as expected. Downloaded 1.19.1 package from Koji https://koji.fedoraproject.org/koji/buildinfo?buildID=1012505. Installed "sudo dnf install mate-power-manager-1.19.1-1.fc27". Reboot. And the 1.19.1 doesn't increases the cpu load. Confirmed.

raveit65, 2018-04-29:
Reverting
https://github.com/mate-desktop/mate-power-manager/commit/6ecb73c1a876e48d2d2ef05603087bd83ac7022f
and
https://github.com/mate-desktop/mate-power-manager/commit/bff61989d0355ee61120ce5d0121d0c99f693569
fixes the high cpu load, but that brings back another issue.
https://github.com/mate-desktop/mate-power-manager/issues/238
which was fixed by
https://github.com/mate-desktop/mate-power-manager/pull/240

lukefromdc, 2018-04-29:
Question then is why does using that GtkImage lead to high CPU use? Not getting  any warnings from it in panel, though I DO get CPU use on a par with Firefox idling from it!

lukefromdc, 2018-04-30:
I just did some testing with the old code. Using the old cairo code to render the image brought back the old redraw problem but stopped the high CPU use. Adding gtk_widget_hide followed by gtk_widget_show_all in the draw callback fixed the redraw issue but brought back high CPU use. This call must run a LOT, as neither cairo drawing nor loading a GtkImage should be all that heavy. I will not try moving that hide/show sequence to the click callback, so it only runs when the icon is actually being changed.

lukefromdc, 2018-04-30:
Moving show/hide to the callback that runs when the icon is actually clicked worked: This cut my total CPU use in half with a bunch of Firefox tabs open, and the icon renders clean and redraws clean.

https://github.com/mate-desktop/mate-power-manager/pull/254

raveit65, 2018-04-30:
@pyaazkaparatha 
Can you please test this build for f27 which include the fix for the issue.
https://koji.fedoraproject.org/koji/taskinfo?taskID=26670644
Pull request works well for me.

pyaazkaparatha, 2018-04-30:
I did and yes the above 1.20.1-2 did lowered the CPU usage.

closed 2018-04-30 by lukefromdc

raveit65, 2018-05-01:
@pyaazkaparatha 
I just pushed the fix to fedora repos.
https://bodhi.fedoraproject.org/updates/FEDORA-2018-c5846e93b3