mate-desktop/mate-power-manager #253: MATE persists to activate automatically the keyboard background light.

Ricky-Tigg, 2018-04-23:
#### Expected behaviour: 
A tool related to laptop's **keyboard background light** adjustment to be present. 

#### Actual behaviour: 
Keyboard background light is automatically set on and furthermore to its maximal intensity level, despite it is currently set in _Gnome_ (installed by default) as disabled. Since _MATE_ lacks property dedicated to  command keyboard background light, disabling this behaviour is not allowed.

Light's intensity cannot be adjusted from keyboard. The issue is not tied to _MATE_; it occurs on any Linux distributions, and then on _Gnome_ and _XFCE_ as well. But only on _Gnome_ it can be adjusted in a power management tool. Currently on physical keyboard, the combination of keys **Fn** (kept pressed down) **+ Right-arrow  + Right-arrow** turns off the background light . While that combination has been applied once, MATE persists systematically to reactivate the light.

#### MATE general version: 1.20.1
#### Package version:  mate-power-manager.x86_64 1.20.1-1.fc28 @updates-testing
#### Linux Distribution: Fedora 28
#### [Link](https://bugzilla.redhat.com/) to downstream report of your Distribution

artebin, 2018-06-17:
I have a similar problem, keyboard backlight is set to 100pc when mate-power-manager starts and when a change is detected e.g. switching battery to AC, leaving idle etc. However the manual change of the keyboard backlight brightness (increase/decrease brightness value) is done properly.
It is just a little bit annoying to see the keyboard backlight 100pc after login when mate-power-manager starts.
I am running xubuntu1804 and working on Dell Inspiron 7737 and MacBook Air7.1.

As far as I can see, others power managers seem not affected, I tried xfce4-power-manager and gnome-power-manager.

I tried to debug a little bit and it seems that when retrieving the brightness value stored in the settings, for example:
value = g_settings_get_int (backlight->priv->settings, GPM_SETTINGS_KBD_BRIGHTNESS_ON_AC);
in gpm-kbd-backlight.c
the value is always 100pc, maybe there is something to fish here.

Ricky-Tigg, 2018-06-17:
Because a user-friendly use matters significantly more than the amount of features provided, in my case MATE has been left unused and will remained as so until a release of a fix.

mdmayfield, 2019-11-25:
I have been working around this issue by doing this:

- In `/etc/dbus-1/system.d/org.freedesktop.UPower.conf`, change `<allow send_destination="org.freedesktop.UPower" send_interface="org.freedesktop.UPower.KbdBacklight"/>` to `deny`.
  - This keeps the keyboard backlight from constantly turning itself on randomly.
  - Note that there are two similar entries; read carefully. On my install the correct one was near the end.

If possible I'd prefer a proper fix, of course.

PR #341 (2020-05-25): gpm-kbd-backlight: do not change keyboard back-light when disabled
Automatically change keyboard back-light only when enabled on battery. Respect `org.mate.power-manager kbd-backlight-battery-reduce` settings.
Keep previous brightness level between state change.

Fixes https://ubuntu-mate.community/t/keyboard-light-keeps-turning-on-after-login-and-or-unlock/6914/24

Closes #253
closed 2020-05-25 by raveit65