mate-desktop/mate-power-manager #257: Support for laptop with multiple batteries

phocean, 2018-05-09:
#### Expected behaviour

Some laptops, like recent Thinkpad (like T460), have two batteries (one internal and one that can be swapped).

The user expects the system to report the global remaining power when on battery (works like this in some different DE).

#### Actual behaviour

Mate-power-manager sees two batteries and report them individually.
The user gets only the remaining time reported by the primary battery.

[image]

#### MATE general version
1.20.1

#### Linux Distribution

Ubuntu 18.04

lukefromdc, 2018-05-09:
Does anyone here have such a laptop? I do not.

alexarnaud, 2018-05-11:
Maybe @vkareh or @flexiondotorg could provide you an answer on Ubuntu Mate indicator.

Best regards,
Alex.

gpsboomerang, 2019-09-22:
I also happen to have a ThinkPad T460s, (with Linux Mate 19.04). Something I've noticed which may help. I suspect the hover pop-up box (see attached) and battery details probably not being updated if it's not the battery being used. ie, with 42% I expect ~2 hours, not 15 minutes.
[image]

gpsboomerang, 2019-09-22:
Looking at the Battery Details page a bit longer, I suspect the battery charge rate (29.8W) is being used for the remaining time calculation. Anyone able to confirm this?

[image]

gpsboomerang, 2019-09-22:
BTW, What does the field 'Supply' mean on the above page? Because it indicates 'Yes' for batteries and AC.

cwendling, 2019-09-24:
> BTW, What does the field 'Supply' mean on the above page? Because it indicates 'Yes' for batteries and AC.

It indicates whether this device is supplying power to the running computer.  Apparently devices that provide power to other connected devices could appear here (imagine the battery of a connected phone or so).

All these info come from upower AFAIK.