mate-desktop/mate-power-manager #258: Implement power lid-close-suspend-with-external-monitor

philthynz, 2018-05-18:
#### Expected behaviour
Laptop does not sleep when connected to external display and the screen closed.

#### Actual behaviour
Laptop goes to sleep

#### Steps to reproduce the behaviour
Plugin in external display. Close laptop lid

#### MATE general version
1.20

#### Linux Distribution
Ubuntu mate 18.04

Can `power lid-close-suspend-with-external-monitor` be implemented please. It's impossible to effectively use an external display with the laptop lid closed.

philthynz, 2018-05-23:
This is a feature request. It belongs to mate-power-manager

Other issue: https://github.com/mate-desktop/mate-desktop/issues/318

spurgelaurels, 2019-04-30:
Vote. 

Same behaviour on Arch. External monitor plugged in and working, but closing laptop puts system to sleep. 

```~ » gsettings get org.mate.power-manager button-lid-ac 
'nothing'
```

```
/etc/udev/rules.d » cat 98-suspend-lid.rules                                    daphne@papilionem
SUBSYSTEM=="power_supply", ENV{POWER_SUPPLY_ONLINE}=="0", RUN+="/usr/bin/powersaving battery"
SUBSYSTEM=="power_supply", ENV{POWER_SUPPLY_ONLINE}=="1", RUN+="/usr/bin/powersaving AC"
```

spurgelaurels, 2019-04-30:
Found an interesting work-around. In the Mate monitors applet, disabling the laptop monitor turns it off until you unplug the HDMI cable, when the display resumes on the laptop. Unfortunately, closing the laptop still puts it to sleep.

philthynz, 2019-04-30:
I gave up on this and stared using ARandR to create a script to swap around monitors. Then launch it with a shortcut key. When I close my laptop lid, I just press superkey + 1 to map back my work monitors. In the Mate power settings, I said close laptop lid do nothing on AC power.

raveit65, 2019-05-03:
I hope someone from the team will start on working this.
For the moment you can use this simple script as workaround for a panel-launcher to switch the gsettings keys on/off, before you close the lid.
Note, the script switched between `suspend` and `nothing` if using gsettings keys for ac and battery, when you use other value than suspend you need to edit it.
```
#!/bin/bash

state_before1=`gsettings get org.mate.power-manager button-lid-ac`
state_before2=`gsettings get org.mate.power-manager button-lid-battery`

echo "previous value on ac was $state_before1"
echo "previous value on battery was $state_before2"
value1="'suspend'"
value2="'nothing'"

if [ "$state_before1" == "$value1" ] || [ "$state_before2" == "$value1" ]; then
       gsettings set org.mate.power-manager button-lid-ac nothing &&
       gsettings set org.mate.power-manager button-lid-battery nothing
  elif [ "$state_before1" == "$value2" ] || [ "$state_before2" == "$value2" ]; then
       gsettings set org.mate.power-manager button-lid-ac suspend &&
       gsettings set org.mate.power-manager button-lid-battery suspend
  else
       echo "previous value was not suspend or nothing"
fi

state_after1=`gsettings get org.mate.power-manager button-lid-ac`
state_after2=`gsettings get org.mate.power-manager button-lid-battery`

echo "current value on ac is $state_after1" &&
echo "current value on battery is $state_after2"
```

BTW. This was done fast and i am sure this can be improved ;)