mate-desktop/mate-power-manager #259: Battery icon does not indicate when charging

mfunder, 2018-07-16:
#### Expected behaviour

Previously, the indicator icon would show a lightning icon while charging, and when fully charged, to indicate the presence of AC power.

#### Actual behaviour
Lightning icon never shows. However, the icon is correctly represented when clicking on the indicator (which shows a list of connected battery devices)

#### Steps to reproduce the behaviour

#### MATE general version
18.04

#### Package version
1.20.1-2ubuntu1

#### Linux Distribution
18.04

#### Link to downstream report of your Distribution

lukefromdc, 2018-07-16:
I too have seen this, suspect an issue with the icon not being redrawn

mfunder, 2018-07-18:
I think this only occurs when fully charged (battery > 99%)

lukefromdc, 2018-07-26:
I just found that mate-power-manager is not redrawing its icon at all, at least not when charging. Charged the laptop's battery, and the battery indicator not only did not show it was charging, the level never changed. Found it at 100% a while later via the tooltip (which did update) and went to the preferences dialog, turned showing the icon off and back on and it then showed charging, with 100% charge.

lukefromdc, 2018-07-26:
On running m-p-m from terminal on the laptop, I got this warning:
`
(mate-power-manager:6185): Gdk-CRITICAL **: 02:48:28.019: gdk_window_thaw_toplevel_updates: assertion 'window->update_and_descendants_freeze_count > 0' failed`

If I changed the icon theme, the icon refresh callback for that must have run, as the new icon correctly reflected charging/discharging. Not only that, from that point forward, with or without changing icon themes back, the icon responded to changing from charging to discharging or vice-versa. My guess is that warning has something to do with it.

We could run the theme changed callback that resets the icon once at applet startup as a workaround, but first we should find and fix that warning and see if that fixes this

raveit65, 2018-08-14:
I have a lightning icon with vanila m-p-m-1.21.0 on fedora 28 and mate-icon-theme.
But i can confirm that the icon doesn't change from 98%  back to 100% state without enable/disable the icon in preferences.
With latest changes on master this is fixed for me.
Probably https://github.com/mate-desktop/mate-power-manager/commit/d5d6e8cd07bf2b15defb01d355d48702c16d23f1

monsta, 2018-08-15:
> Lightning icon never shows. However, the icon is correctly represented when clicking on the indicator (which shows a list of connected battery devices)

I think we're talking about two different issues here...

First, we have a redraw/refresh issue which is apparently fixed in the mentioned commit. It's also discussed in https://github.com/mate-desktop/mate-power-manager/issues/260.

Second, we have a patched mate-power-manager in Ubuntu MATE 18.04. It won't show the tray icon at all, and the power preferences UI won't let you change the icon showing preference anymore.

https://salsa.debian.org/debian-mate-team/mate-power-manager/commit/ec4c7b4589be76fa4afa2454424de40caf95d5bf
https://salsa.debian.org/debian-mate-team/mate-power-manager/commit/d0cbb8fb57716b6be89eddef6d3c0d4767f4e0c3
https://salsa.debian.org/debian-mate-team/mate-power-manager/commit/c97b38c42f3d792f785be35eb026deb7fde4f2d7

It's done because `indicator-power` is supposed to handle it. All the panel layouts in Ubuntu MATE 18.04 have indicators enabled by default. I guess you're not supposed to remove them if you wish to customize some layout...

@mfunder: so, if you see the described behavior, it's intended in the distro.

raveit65, 2018-08-15:
Any way, fix for our applet is in 1.20 branch d5d6e8cd07bf2b15defb01d355d48702c16d23f1
@monsta 
You can close this report if it is an issue with ubuntu/debian stuff, or not?

monsta, 2018-08-16:
Yes, just wanted a bit more confirmation as `indicator-power` doesn't show anything on the desktop for me. Thanks to @vkareh for confirming. Closing as Ubuntu-specific thing.

closed 2018-08-16 by monsta

vkareh, 2018-08-16:
Correct. The OP seems to experience this issue with the indicator-power, and most likely with the defauly Ambiant-MATE theme... when changing to any other theme I can see the icon with the correct status (lightning, plug thing)