mate-desktop/mate-power-manager #260: pushed by accidental

raveit65, 2018-08-13:
I pushed a few commit to master without doing a PR.
Can you please review/test those?
https://github.com/mate-desktop/mate-power-manager/commit/c569240f040f7b4394336569c803f58bd85f6b01
https://github.com/mate-desktop/mate-power-manager/commit/d5d6e8cd07bf2b15defb01d355d48702c16d23f1
https://github.com/mate-desktop/mate-power-manager/commit/71eb5487cb205d5e637439305232f4bad626a04a
https://github.com/mate-desktop/mate-power-manager/commit/b3c86df1178c9f64548554d9ba5f0a721560a230
@monsta @lukefromdc @sc0w @vkareh

lukefromdc, 2018-08-13:
I will test this later today

raveit65, 2018-08-14:
> Not only did it work, the battery icon now responds to change of state between charging and discharging. 

I can confirm that. Backporting to 1.20 ?

monsta, 2018-08-14:
Cool. It should be indeed backported then.

raveit65, 2018-08-14:
I will test it first as patch with a 1.20.x tarball.

raveit65, 2018-08-14:
Ok, https://github.com/mate-desktop/mate-power-manager/commit/d5d6e8cd07bf2b15defb01d355d48702c16d23f1 with latest 1.20.x release fixes the issue too.
So, it makes sense to backport the commit.

vkareh, 2018-08-15:
These all work for me. I can see the battery tray icon responding to charge/discharge events :+1: 
Brightness applet also works as expected.

monsta, 2018-11-25:
Ok... should we close this as tested?

lukefromdc, 2018-11-25:
OK by me

raveit65, 2018-12-17:
gpm-button: replace deprecated Gdk functions is already in 1.20 branch, closing.

closed 2018-12-17 by raveit65