mate-desktop/mate-power-manager #261: mate-power-backlight-helper executes really slowly

kumanoko24, 2018-09-13:
#### Expected behaviour
When press `fn + brightness-increase` keys on keyboard, the brightness UI of the mate-power-manager should pops up immediately.
Or when execute `sudo pkexec mate-power-backlight-helper --set-brightness N`, it should immediately change the brightness.

#### Actual behaviour
When press `fn + brightness-increase` keys on keyboard, the brightness UI of the mate-power-manager pops up after 5 - 10 seconds then changes the brightness.
Or when execute `sudo pkexec mate-power-backlight-helper --set-brightness N`, it changes the brightness after 5 - 10 seconds, with a 100% usage of one cpu thread.

#### Steps to reproduce the behaviour
As above. I am using Thinkpad T460p.

#### MATE general version
Latest Ubuntu-MATE 18.04

#### Package version
```
Package: mate-power-manager
Version: 1.20.1-2ubuntu1
Priority: optional
Section: universe/x11
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian+Ubuntu MATE Packaging Team <debian-mate@lists.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 467 kB
Depends: default-dbus-session-bus | dbus-session-bus, mate-notification-daemon | notification-daemon, mate-power-manager-common (= 1.20.1-2ubuntu1), policykit-1, systemd | consolekit, upower, libc6 (>= 2.27), libcairo2 (>= 1.10.0), libcanberra-gtk3-0 (>= 0.25), libcanberra0 (>= 0.2), libdbus-1-3 (>= 1.9.14), libdbus-glib-1-2 (>= 0.88), libgdk-pixbuf2.0-0 (>= 2.22.0), libglib2.0-0 (>= 2.37.3), libgnome-keyring0 (>= 2.20.3), libgtk-3-0 (>= 3.21.4), libmate-panel-applet-4-1 (>= 1.6.1), libnotify4 (>= 0.7.0), libpango-1.0-0 (>= 1.22.0), libpangocairo-1.0-0 (>= 1.14.0), libupower-glib3 (>= 0.99.0), libx11-6, libxext6, libxrandr2 (>= 2:1.2.99.2)
Suggests: mate-polkit
Homepage: http://www.mate-desktop.org/
Task: ubuntu-mate-core, ubuntu-mate-desktop
Supported: 3y
Download-Size: 118 kB
APT-Manual-Installed: yes
APT-Sources: http://tw.archive.ubuntu.com/ubuntu bionic/universe amd64 Packages
Description: power management tool for the MATE desktop
 MATE Power Manager is a session daemon for the MATE desktop
 that takes care of system or desktop events related to power, and
 triggers actions accordingly. Its philosophy is to completely hide
 these complex tasks and only show some settings important to the user.
 .
 The MATE power manager displays and manages battery status, power plug
 events, display brightness, CPU, graphics card and hard disk drive
 power saving, and can trigger suspend-to-RAM, hibernate or shutdown
 events, all integrated to other components of the MATE desktop.

```

#### Linux Distribution
Ubuntu 18.04 (Ubuntu-MATE)

raveit65, 2018-09-13:
I can't confirm that with fedora 28. Looks like an Ubuntu issue
Why did you didn't inform your ubuntu maintainer with filing first out a report there?

memeplex, 2019-02-20:
I've been this (pkexec running too slowly) on Manjaro Gnome Edition.

https://forum.manjaro.org/t/very-slow-response-of-brightness-controls-in-gnome/74891/6