mate-desktop/mate-power-manager #265: Limit CPU for non-focused applications

brimston3, 2018-10-31:
Feature req. for limiting CPU time (eg. cgroups cpu.cfs_quota_us/cpu.cfs_period_us) for GUI applications that do not have focus and their children. When an application loses focus, or leaves the top of the stack, it gets placed under a global shared execution cap of X% (say 1%). When it regains focus/top of stack, the execution cap is removed. This is basically how mobile OSes treat background applications to preserve battery life.

Use case is for laptops, but I can imagine it makes other programs more responsive. Exceptions would have to be made for things like toolbars.

In concrete example, on a laptop working on a l.o. calc spreadsheet, I open a javascript-heavy webpage in chromium, then switch back. I'd like the chromium processes and any other background GUI tasks to be CPU execution capped to 1% total while I continue working on my spreadsheet.

lukefromdc, 2018-10-31:
This would HAVE to be user-selectable and is something I would always
disable. I routinely let something like vdeo editing run in background while
I do something else an no way I would cap it due to not being focussed

vkareh, 2018-10-31:
I don't even think this should be part of the DE... Tinkering with CPU priority levels seems like the type of thing we should let a much lower-level system worry about, maybe even at the kernel level, but not the DE.... Are there any other desktops that do this?

closed 2018-10-31 by lukefromdc

brimston3, 2018-10-31:
I don't think it's possible to do right now on any DE. It needs cooperation from multiple levels and it'd only work on linux. At least the WM or X server needs to be involved because it's the only one that knows the current focus/window stack. Whatever spawns new processes would also have to be aware of the goal to assign new processes to the appropriate cgroup pool, and something would have to initialize those pools too. I understand that this is not the right place for this kind of suggestions and would appreciate direction to the right place for it.

lukefromdc, 2018-10-31:
I have no idea at all but suspect it would have to include the kernel. Would have to be 
selectable by users as many would consider it an anti-feature. I still remember Windows 3.1, 95
et all being slagged as poor at multitasking compared to Apple at that time to the point that 
the Washington Post ran a "Windows 2000= Mac 1984" cartoon. We don't want to ever force
users to throttle back heavy background processes they deliberately start such as video rendering,
and I would never run a kernel or any other program that forced this.

vkareh, 2018-11-01:
Would something like systemd have access to change nice values? If such an API is exposed we could use dbus to interact with it, assuming such an API exists