mate-desktop/mate-power-manager #274: Problem compiling mate-power-manager from source - fails at "undefined reference to `up_client_enumerate_devices_sync'"

rubyFeedback, 2019-02-08:
Hello mate-folks,

I am trying to compile:

mate-power-manager-1.21.1

from the official tarball-source.

I use this way to try this:

    ./configure --prefix=/usr --without-keyring

(gnome-keyring gave me some unrelated problems, so I am trying to
just get mate-power-manager installed.)

An error occurs lateron. The error is:

---

  CCLD     mate-power-manager
/System/Index/bin/ld: mate_power_manager-gpm-engine.o: in function `gpm_engine_coldplug_idle_cb':
/Depot/jjj/mate-power-manager-1.21.1/src/gpm-engine.c:855: undefined reference to `up_client_enumerate_devices_sync'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:1100: mate-power-manager] Error 1
make[3]: Leaving directory '/Depot/jjj/mate-power-manager-1.21.1/src'
make[2]: *** [Makefile:945: all] Error 2
make[2]: Leaving directory '/Depot/jjj/mate-power-manager-1.21.1/src'
make[1]: *** [Makefile:483: all-recursive] Error 1
make[1]: Leaving directory '/Depot/jjj/mate-power-manager-1.21.1'
make: *** [Makefile:415: all] Error 2

---

I am not entirely sure what up_client_enumerate_devices_sync is; googling
didn't give me a lot of too terribly useful information, but I think
either way perhaps the build system could detect or check for this
problem if possible. Anyway, I am reporting this here in the event that
it has not been reported yet. Feel free to close this issue at any moment
in time.

Here is the log:

checking for a BSD-compatible install... /System/Index/bin//install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /System/Index/bin//mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether UID '0' is supported by ustar format... yes
checking whether GID '0' is supported by ustar format... yes
checking how to create a ustar tar archive... gnutar
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /System/Index/bin//sed
checking for grep that handles long lines and -e... /System/Index/bin//grep
checking for egrep... /System/Index/bin//grep -E
[... 664 lines cut ...]
make[3]: *** [Makefile:1100: mate-power-manager] Error 1
make[3]: Leaving directory '/Depot/jjj/mate-power-manager-1.21.1/src'
make[2]: *** [Makefile:945: all] Error 2
make[2]: Leaving directory '/Depot/jjj/mate-power-manager-1.21.1/src'
make[1]: *** [Makefile:483: all-recursive] Error 1
make[1]: Leaving directory '/Depot/jjj/mate-power-manager-1.21.1'
make: *** [Makefile:415: all] Error 2
Making install in src
make[1]: Entering directory '/Depot/jjj/mate-power-manager-1.21.1/src'
make  install-am
make[2]: Entering directory '/Depot/jjj/mate-power-manager-1.21.1/src'
  CCLD     mate-power-manager
/System/Index/bin/ld: mate_power_manager-gpm-engine.o: in function `gpm_engine_coldplug_idle_cb':
/Depot/jjj/mate-power-manager-1.21.1/src/gpm-engine.c:855: undefined reference to `up_client_enumerate_devices_sync'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:1100: mate-power-manager] Error 1
make[2]: Leaving directory '/Depot/jjj/mate-power-manager-1.21.1/src'
make[1]: *** [Makefile:2090: install] Error 2
make[1]: Leaving directory '/Depot/jjj/mate-power-manager-1.21.1/src'
make: *** [Makefile:483: install-recursive] Error 1

My host system is a heavily modified slackware system as base. I
compiled most things there from source; latest binutils,
gcc 8.2.0 etc... only glibc and the linux kernel is quite old, but
I don't think it is related to the above error.

monsta, 2019-02-09:
@willysr: since it's a Slackware, maybe you can help?

willysr, 2019-02-09:
It's pretty hard to help when he is using a custom installation as i normally support full installation only and tbh, i haven't tried building 1.21.x yet.

Did we still support legacy upower? I'm suspecting that was the cause

willysr, 2019-02-10:
UPower 1.0 API removed up_client_enumerate_devices_sync:
https://lists.freedesktop.org/archives/devkit-devel/2013-October/001519.html

monsta, 2019-02-10:
Legacy upower is still supported (actually, only because of Slackware which still has 0.9.23).
We have some `#if`s for that. In `gpm_engine_coldplug_idle_cb` there's this piece of code:

```
#if !UP_CHECK_VERSION(0, 99, 0)
	/* get devices from UPower */
	ret = up_client_enumerate_devices_sync (engine->priv->client, NULL, &error);
	if (!ret) {
		egg_error ("failed to get device list: %s", error->message);
		g_error_free (error);
		goto out;
	}
#endif
```
So this part should not be built with older upower.

@rubyFeedback: please check your upower version.