mate-desktop/mate-power-manager #282: Unable to scroll interface tabs with mouse wheel on MPM

N0rbert, 2019-05-13:
#### Expected behaviour
On Ubuntu MATE 16.04 LTS user was able to use mouse wheel to scroll interface tabs.

#### Actual behaviour
On development version of Ubuntu 19.10 this is not possible - tabs do not react on mouse wheel.

#### Steps to reproduce the behaviour
1. Install Ubuntu MATE 19.10
2. Open any of `mate-power-preferences`, `mate-power-statistics`
3. Try to switch interface tabs with mouse wheel

#### MATE general version
1.22.1

#### Package version
1.22.1-0ubuntu1

#### Linux Distribution
Ubuntu 19.10

#### Link to downstream report of your Distribution
https://bugs.launchpad.net/ubuntu/+source/mate-power-manager/+bug/1828802

raveit65, 2019-05-13:
Was this behavior intended?

N0rbert, 2019-05-13:
As long time GNOME2 user this functionality is intended for me. Seems to be GTK3-caused regression.

tamplan, 2019-07-11:
Issue resolved by https://github.com/mate-desktop/mate-power-manager/commit/67b0883a92607d52ece6ab7e8f8869e7f55bab29 merged.

closed 2019-07-15 by vkareh