mate-desktop/mate-power-manager #302: Display can not be totally dimmed when running Toshiba Satellite L650 D

JanuszChmiel, 2019-09-23:
#### Expected behaviour
When I run mate-power-preferences binary module, when I position The slider to 0, display should be totally dimmed, it should not produce light at all.

#### Actual behaviour
When I set The slider to 0, display still produces some light. Next problem. Probably ATI firmware module or other kernel part dim display after several minutes of inactivity even if I set The slider dim The display to never.

#### Steps to reproduce the behaviour
Run mate-power-preferences
Set The slider to 0.
Set The next slider dim display to never.
Uncheck The checkbox dim display when idle.
Press The close button.

#### MATE general version
1.22.1

#### Package version

#### Linux Distribution
Fedora 31 ARCH 64 Mate spin.

#### Link to downstream report of your Distribution
I did not filled it out.
Important remark
Display is correctly dimmed when I use modern EFI based computers such as Lenovo G50 series. The informatics question is, if is it possible to develop algorithm, which will dim The display totally also on MBR based BIOSES. But. When some other process can really dark The display, there is some way to do that at The software level. I Am ready to constructively participate to make this feature work for Toshiba Satellite L650 D systems. I do not see at all so I want to prevent lamp and condensator which produce high energy from burn out. I do not need light at all and really, it have worked for Lenovo G50. I know, that it is not easy programmers task. But yours team is full of amazing and kind C language specialists. This is The reason, why Mate is so stable, reliable and for now, the only one GTK based really fully accessible desktop environment for visually impaired. No memory hungrymodules, which allocates 320 MB or more after short time of activity. Very fast responsiveness in The classic Mate panel menu.
Thank you very very much for yours kind access to my plea.

raveit65, 2019-09-23:
> When I run mate-power-preferences binary module, when I position The slider to 0, display should be totally dimmed, it should not produce light at all.

Slider to zero doesn't mean the monitor is dark or the monitor is powered off.
Or what you mean with `totally dimmed`?
But you can suspend the monitor after a some idle time.

> Probably ATI firmware module or other kernel part dim display after several minutes of inactivity even if I set The slider dim The display to never.

This is upower or systemd. You need to disable or configure their settings.
You will see that behaviour when you terminate mate-power-manager or start a MATE session with mate-power-manager disabled in autostart.

JanuszChmiel, 2019-09-23:
Really. My sighted mother has confirmed Me, that when I have positioned The slider to zero on my Lenovo laptop, screen have been powered off. It did not produce light. Itwas dark like when I did not power The computer.
I have liked this feature very much. I will try this behaviour on Lenovo laptop to be 100 % sure that somebody of yours team did not recode source codes so this feature has disappeared. Thank you very very much for yours patience with Me.

JanuszChmiel, 2019-09-23:
So really. Lenovo G50 laptop really power display off when slider is set to zero. So it is my desired random bonus may be that from Kernel driver developer or from The manufacturer of display. But I know, that it would be probably too complex try to let monitor disabled and let user work with mouse and keyboard at The same time. I will accept The sad fact, that no every driver for graphical chip can allow this behaviour default. Debian accessibility developers have tried to create some instructions how to make display turned off and enabling users to work at The same time. But their tip do not work on every system. So it is not easy to make such thinks universal.
Thank you very much for yours time, for yours understanding. And thank you again for yours 100 % accessible GUI. In todays world it is not standard. All mate modules, which creates GUi are also very responsive while using them with Orca screen reader. This is The reason why I have so deep appreciation to all C developers. And whole system is very stable. Including power-manager and all dependent modules. I wish you much happynes and much achievements while you are programming.

raveit65, 2019-09-23:
> So really. Lenovo G50 laptop really power display off when slider is set to zero.

This is definitely wrong.
How do you have control about your desktop when this happens?
Btw. what do you do to bring back normal desktop?

vkareh, 2019-09-23:
@raveit65, this is common for blind users. I have a friend that powers off his monitor and so his battery lasts for days. He uses Orca to interact with the laptop, since the screen means nothing to him.

raveit65, 2019-09-23:
Sorry, Looks like my english is too bad and i over read that very unclear information.
Ok , than our accessibility boss should help...
@alexarnaud

JanuszChmiel, 2019-09-24:
I can control my desktop perfectly. And monitor can be turned on when I set The number greater than zero. When monitor is turned off and slider is set to zero, I can do all tasks without crashes. Because I do not see at all I only use Orca screen reader. This is The reason, why I would like to have The feature which would power monitor totally of as a official feature of Mate desktop environment. This feature exist for Android Samsung mobile phones. I know, that from yours developers perspective behaviours of mate-power-preferences on The Lenovo laptop is wrong. Because sightedd user would not have The chance to operate The computer.
The question is, if The standalone feature represented by checkbox turn monitor of while working with The system (for visually impaired users of Orca only) would be implemented.
To be honest, I Am glad, that for Lenovo laptop slider turn The monitor off. So even it can be see as bug for sighted users, for Me, it is positive feature. 
Thank you very much for yours understanding.

mhussaincov, 2019-09-24:
hi there,
orca user here,
this would be a great setting to have, maybe as a checkbox instead of a slider?
people who have no site don't require the screen to be on, it would
save on energy bills also,
Majid

vkareh, 2019-09-24:
@JanuszChmiel, my Dell XPS turns the screen off with the slider all the way down, but I think I added some kernel parameters to change screen brightness behavior. That was a while ago and I don't remember entirely.

In any case, the mate-display-properties allows you to have a status icon with monitor properties, and I think one of the options there is to turn the screen off. Would this be a suitable alternative in the meantime?

JanuszChmiel, 2019-09-24:
Yes, it would be 100 % amazing. I appreciate you that you try to solve my opened issue with so kind access to my plea. When you will have functioning prototype, please let Me know, I will try to even compile The corresponding source code to test it before Fedora team will add changed module to Their repository.

alexarnaud, 2019-09-24:
The @vkareh proposal seems reasonable to me.

FYI, there is a section dedicated to turn off the screen monitor on the Debian accessibility wiki: https://wiki.debian.org/accessibility#Screen_off

Note that such feature to disable the screen are not available natively on Android or Mac, those are integrated as screen reader feature.

Best regards.

vkareh, 2019-09-24:
Oh, no, it wasn't a proposal, it was a question. Currently the
mate-display-properties tool has a checkbox to "Show monitors in panel".
This enables a status icon that, when activated, has an option to turn
the monitor off. My question is whether this works for your use-case?

On a side note, I just tested my Lenovo P50 and it _also_ turns the
screen off when the brightness is set all the way down.

JanuszChmiel, 2019-09-24:
I AM very sorry for my misunderstanding. I will try The function. If it will turn The monitor off or no. I will let you know.

vkareh, 2019-09-24:
Thanks, sorry for the confusion.

So far it seems like an issue with that particular hardware. Both my
laptops turn off the display backlight entirely, and I don't have any
other hardware to test this with.

If there are other laptops that have this issue, we should make a list
and see what the pattern is.

JanuszChmiel, 2019-09-24:
I do not want to bother yours team so much. But I can not access icon context menu for monitors from Mate panel. I can have only access to The one button show desktop. When I press CTRL+ALT+TAB. I can make some C language experiment to inject The procedure for turning monitor off only, but it will be not easy mission since I Am not so big professional like C language programmers here. It would be possible to create little binary module which routine for turning monitor off only. And user could assign hod key from Mate control center for calling this little app. If monitor will not be turned off, it is very probably hardware problem. For now, I will kind ask my kind and positive human being my mother so she can click me with The right mouse button on The monitors icon and we will try to turn monitor off by using her assistance.

I have important information. This model of Toshiba laptop contain display with very strong lamp it have special manufacturer assignment and it really light very much also for sighted users. So may be, that it is hardware issue that internal circuit will not enable us to turn display completely off. So may be, that this is not programmable een from kernel level. I Am very glad, that zero value of slider can turn off many other monitors. I will test turn monitor off with my mother from system tray by using context menu.

JanuszChmiel, 2019-09-24:
I have tried to turn The monitor off from The system tray but I have got The message, that this monitor setting can not be used. And monitor light, it did not been turned off. Do you think, that this is The indicator, that VIDEO BIOS or other hardware problem really prevent us from turning this specific display off and that some software work around can not be used? 
I have found out, that when I press CTRL+ALT+Esc, I can got to The panel two and all icons are accessible. I have analysed all mate shortcut keys and really, whole icons area is fully accessible.

JanuszChmiel, 2019-09-24:
When I have set The checkbox dim display when idle to checked on mate-power-preferences settings dialog box and when I have set The slide turn display after 1 minute, so not never, monitor have been really turned off after one minute of inactivity. So in this case, routine really disable The monitor. But I AM afraid, that this routine build in to this algorithm can not be used to have monitor turned off and working with The system at The same time. But we are near to The solution.
I have also turned The screen saver off to be sure that really power-preferences app can do this magic.