mate-desktop/mate-power-manager #305: Power Statistics cannot enable timerstats in Wakeups tab

flexiondotorg, 2019-09-26:
#### Expected behaviour

Power Statistics for the Processor should output Wakeup stats in the table.

#### Actual behaviour

The table is empty and Processor wakeups per second says `GDBus.Error:org.freedesktop.UPower.GeneralError: cannot enable timerstats`.

#### Steps to reproduce the behaviour

Open Power Statistics and click Processor.

#### MATE general version

1.22.2 and git master

#### Package version

1.22.2-0ubuntu2

#### Linux Distribution

Ubuntu MATE 19.10

#### Link to downstream report of your Distribution

None as yet.

flexiondotorg, 2019-09-26:
This issue was **_not_** caused by #303. I tested 1.22 and git master with and without that patch applied.

flexiondotorg, 2019-09-26:
I have just tested Ubuntu MATE 19.04 with `mate-power-manager` 1.20.3-2 and `upower` 0.99.10-1 and the same issue is present.

raveit65, 2019-09-26:
I can confirm the issue with released 1.22.2/1/0 and upower-0.99.10/9.

rbuj, 2019-09-26:
CONFIG_TIMER_STATS was removed from Kernel https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?h=v4.19-rc6&id=20dcfe1b7df4072a3c13bdb7506f7138125d0099

raveit65, 2019-09-26:
Now i know why gnome-power-statistic doesn't display it any more, only my mouse device.
[image]

raveit65, 2019-09-26:
So we need that?
https://gitlab.gnome.org/GNOME/gnome-power-manager/commit/8621440250db093f7ec614193126ada1f2eedd9e
With a check for kernel-version?

flexiondotorg, 2019-09-27:
Probably just drop it. Eventually there will be no kernels that support the feature.

Question is, what UI changes are required? On my desktop and VMs the only thing I see in Power Statistics is the broken time stats.

lukefromdc, 2019-09-28:
This is confirmed here too, on kernel 5.3.0 and I concur that if it depends on a no longer avaliable kernel feature we should drop it.

 GNOME did manage to find another way of doing this (at least with 2018 kernels) in their gnome-usage application if anyone wants to find a way to bring it back-or fork that whole application and port it to the MATE user interface for that matter.

Ph0rkoz, 2019-10-24:
Yes, broken in latest linux mint. :(

PR #319 (2020-02-24): Remove the CPU wakeups feature
Closes #305

Based on https://github.com/GNOME/gnome-power-manager/commit/8621440250db093f7ec614193126ada1f2eedd9e
closed 2020-02-24 by raveit65