mate-desktop/mate-power-manager #308: Reboot stalled

patricktokeeffe, 2019-09-26:
> Sorry if this was reported already, my searches turned up nothing. 

Sometimes, not always, when user restarts the following "A program is still running" dialog appears:

[image]

If I ignore the dialog and instead press the panel icon to switch modes, then reboot proceeds immediately. 

Observed with Ubuntu MATE 16.04 LTS, with version 1.16.2:
```
lar@lgr-13-0199:~$ apt-cache policy mate-power-manager
mate-power-manager:
  Installed: 1.16.2-1~xenial1.0
  Candidate: 1.16.2-1~xenial1.0
  Version table:
 *** 1.16.2-1~xenial1.0 500
        500 http://ppa.launchpad.net/ubuntu-mate-dev/xenial-mate/ubuntu xenial/main armhf Packages
        100 /var/lib/dpkg/status
     1.12.1-1 500
        500 http://ports.ubuntu.com xenial/universe armhf Packages
```

lukefromdc, 2019-09-27:
That should only be possible with the inhibit applet in the panel and set to "automatic sleep inhibited" which also inhibits reboot for some reason.

patricktokeeffe, 2019-10-01:
Thanks @lukefromdc you are correct inhibiting sleep inhibits reboot too. I can almost appreciate that feature.. however, I find it overall confusing because of the applet's name. 

The applet help would not open for me (*missing en_US translation?*) but I didn't find 'reboot' or 'shutdown' mentioned in the source. Likewise the [upstream docs](https://help.gnome.org/users/gnome-power-manager/stable/applets-general.html.en_GB#applets-inhibit) also do not mention inhibiting reboots.

patricktokeeffe, 2019-10-09:
Hi @raveit65 sorry there's no headers but the op did/does contain all required info