mate-desktop/mate-power-manager #314: Thinkpad /sys/class/power_supply/BAT0/charge_stop_threshold not supported

jankratochvil, 2019-12-25:
#### Expected behaviour
Fully charged status.

#### Actual behaviour
Charging icon
Laptop battery 30 minutes until charged (79.0%)

#### Steps to reproduce the behaviour
echo 80 >/sys/class/power_supply/BAT0/charge_stop_threshold
Discharge under 80%.
Charge (which will stop at 79%).

#### MATE general version

#### Package version
mate-power-manager-1.22.2-1.fc30.x86_64

#### Linux Distribution
Fedora 30 x86_64

#### Link to downstream report of your Distribution
none

jankratochvil, 2020-11-10:
still valid: mate-power-manager-1.24.2-1.fc33.x86_64

raveit65, 2020-11-10:
I don't see this issue with my old MSI fx700 notebook.
Works fine here.
Sorry, i don't have other hardware.

jankratochvil, 2020-11-11:
Do you really have battery state of charge under 100% and no longer charging?
```
$ cat /sys/class/power_supply/BAT0/charge_stop_threshold
81
$ acpi
Battery 0: Unknown, 80%
```