mate-desktop/mate-power-manager #321: some status icons have wrong size

sunweaver, 2020-02-14:
#### Expected behaviour

icons in folders named <width>x<height> are of size <width>x<height>

#### Actual behaviour

Some of the status icons are all 48x48 in all icon size folders.

W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/128x128/status/gpm-keyboard-020.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/128x128/status/gpm-keyboard-040.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/128x128/status/gpm-keyboard-080.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/128x128/status/gpm-mouse-020.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/128x128/status/gpm-mouse-040.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/128x128/status/gpm-mouse-080.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/128x128/status/gpm-phone-020.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/128x128/status/gpm-phone-040.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/128x128/status/gpm-phone-080.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/256x256/status/gpm-keyboard-020.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/256x256/status/gpm-keyboard-040.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/256x256/status/gpm-keyboard-080.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/256x256/status/gpm-mouse-020.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/256x256/status/gpm-mouse-040.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/256x256/status/gpm-mouse-080.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/256x256/status/gpm-phone-020.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/256x256/status/gpm-phone-040.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/256x256/status/gpm-phone-080.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/64x64/status/gpm-keyboard-020.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/64x64/status/gpm-keyboard-040.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/64x64/status/gpm-keyboard-080.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/64x64/status/gpm-mouse-020.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/64x64/status/gpm-mouse-040.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/64x64/status/gpm-mouse-080.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/64x64/status/gpm-phone-020.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/64x64/status/gpm-phone-040.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/64x64/status/gpm-phone-080.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/96x96/status/gpm-keyboard-020.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/96x96/status/gpm-keyboard-040.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/96x96/status/gpm-keyboard-080.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/96x96/status/gpm-mouse-020.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/96x96/status/gpm-mouse-040.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/96x96/status/gpm-mouse-080.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/96x96/status/gpm-phone-020.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/96x96/status/gpm-phone-040.png 48x48
W: mate-power-manager-common: icon-size-and-directory-name-mismatch usr/share/mate-power-manager/icons/hicolor/96x96/status/gpm-phone-080.png 48x48

#### Steps to reproduce the behaviour

Build the package and check icon sizes in /usr/share/mate-power-manager/icons/hicolor/<X>x<Y> for for X = Y != 48.

#### MATE general version

1.24.0

#### Package version

upcoming 1.24.0-1

#### Linux Distribution

Debian

#### Link to downstream report of your Distribution

raveit65, 2020-02-14:
Probably caused by https://github.com/mate-desktop/mate-power-manager/commit/a9ff56c6de34f8d3db48f3405c5ee671dc281194
@vkareh 
Copy & paste error?

vkareh, 2020-02-14:
Yep, in trying to remember about this, I think we did not have origin SVGs for some of the icons, so when I copied the icon folders to recreate them, those that didn't have SVG just weren't replaced in my script.

I'll see if I can find sources...

sunweaver, 2020-02-14:
@vkareh: this is still an issue for 1.24.0, obviously.

vkareh, 2020-02-14:
Yes, I agree. I see it in my sources in `master` branch. Thanks for reporting :)

vkareh, 2020-02-14:
what would be the best approach here? From a status icon point of view, it's rare that anyone would have a panel with the tray applet larger than 48px. In my case (using Menta theme) if I make the panel 48px, the tray applet wraps the icon to a smaller size. 47px seems to be the largest.

Screenshots:
* 47px [image]

* 48px [image]

Are there any circumstances in which status icons are shown elsewhere (i.e. outside of the tray applet)?

Is there any reason to even have status icons larger than 48px?

Let's say that we do, would it be better to have icons mapped to 48px that just render blurry or to not have the icons at all? Basically, do we remove the 48px icons from the larger size folders or leave them there? This is based on the premise that we don't find the sources (which I feel we won't).

Thoughts?

raveit65, 2020-02-14:
> Are there any circumstances in which status icons are shown elsewhere (i.e. outside of the tray applet)?

The icons are used at 3 places, tested with my wireless mouse.
na-area, menu and mpm-statistics.
[image]

> Is there any reason to even have status icons larger than 48px?

I agree with using a horizontal panel, but with a vertical panel the panel-size can be larger in general. But it seems that the status-icon don't proper scale with larger icon-size. Looks like the size is limited to 48px or smaller and the icon is cut off with vertical panel.
[image]
[image]
[image]
 
Sorry, for German screenshots :smiley: 

PS: i have looked at gnome-power-manager-3.0 git and the SVGs are missing there too.

raveit65, 2020-02-14:
I don't have a low battery with 20% , 40% or 80% charge for testing if the icon is really blurred.

raveit65, 2020-02-14:
First new SVGs for mouse device are at https://github.com/mate-desktop/mate-power-manager/tree/svg-icons

rbuj, 2020-02-14:
All png icons can be created from svg files using rsvg-convert:
```shell
for d in 64 128 256; do for f in data/icons/scalable/status/*.svg; do F=$(basename $f); echo "rsvg-convert -w $d -h $d $f -o data/icons/${d}x${d}/status/${F%.*}.png"; done; done | sh
```

kretcheu, 2020-02-20:
Interesting, IMHO, observation.

If I run mozo at mate-terminal, I can't see icons in folders named, but if a run:
`export LANG=C; mozo`
 all icons are there!

I hope that it can help to solve problem.

closed 2020-02-20 by raveit65