mate-desktop/mate-power-manager #327: mate-power-manager.desktop broken near Categories= and OnlyShowIn fields (startup of m-p-m not limited to MATE desktop)

sunweaver, 2020-02-24:
#### Expected behaviour

an intact mate-power-manager.desktop file in XDG autostart. MATE power manager only starting up in MATE.

#### Actual behaviour

OnlyShowIn= key gets appended to empty Categories= key (missing EOL char). MATE power manager gets launched on all other desktop envs installed on the system.

#### Steps to reproduce the behaviour

Login to non-MATE desktop env, inspect /etc/xdg/autostart/mate-power-manager.desktop (near Categories= and OnlyShowIn= keys).

#### MATE general version

1.24

#### Package version

1.24.0-1

#### Linux Distribution

Debian

#### Link to downstream report of your Distribution

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=951802

raveit65, 2020-02-24:
The issue will introduced during make command because of missing semicolon after port to gettext.
```
[rave@mother mate-power-manager]$ cat data/mate-power-manager.desktop.in.in 
[Desktop Entry]
<cut>
# Translators: Search terms to find this application. Do NOT translate or localize the semicolons! The list MUST also end with a semicolon!
Categories=
OnlyShowIn=MATE;
<cut>
```
The result is `Categories=OnlyShowIn=MATE;`
Adding a semicolon fixes the issue.
```
[rave@mother mate-power-manager]$ cat data/mate-power-manager.desktop.in.in 
[Desktop Entry]
<cut>
# Translators: Search terms to find this application. Do NOT translate or localize the semicolons! The list MUST also end with a semicolon!
Categories=;
OnlyShowIn=MATE;
<cut>
```
The result is:
```
# Translators: Search terms to find this application. Do NOT translate or localize the semicolons! The list MUST also end with a semicolon!
Categories=;
OnlyShowIn=MATE;
```
No idea if this is the right fix or we should simply remove an empty Categories entry.

raveit65, 2020-02-24:
`desktop-file-validate mate-power-manager.desktop` is successful with
```
# Translators: Search terms to find this application. Do NOT translate or localize the semicolons! The list MUST also end with a semicolon!
Categories=;
```
or without the `Categories` entry.

yetist, 2020-02-24:
According to the debian patch, can we delete the `Categories=` item directly?
https://sources.debian.org/patches/mate-power-manager/1.24.0-2/1001_fix-only-showin.patch/

raveit65, 2020-02-24:
Or the fedora patch? https://src.fedoraproject.org/rpms/mate-power-manager/c/c0352696a760435bc3b3aa21ecf7f1d3a14b6a10?branch=master
I think so. Most desktop files in autostart doesn't have `Categories` entry.

raveit65, 2020-02-24:
.....waiting for a pull request from debian maintainer and mate contributer.

vaurelios, 2020-03-31:
Can i make the pull request? (deleting the entry)

raveit65, 2020-03-31:
Sure, seems like @sunweaver is to lazy to help us.

commit 6ff43e0c (2020-04-01): Remove unused entry from autostart

Fix #327.
Remove the unused Categories= entry from
the autostart dekstop file. This empty entry
and no semicolon causes the OnlyShowIn= entry
to be appended to Categories= entry.
commit 05544d4a (2020-04-01): Remove unused entry from autostart

Fix #327.
Remove the unused Categories= entry from
the autostart dekstop file. This empty entry
and no semicolon causes the OnlyShowIn= entry
to be appended to Categories= entry.
closed 2020-04-01 by raveit65