mate-desktop/mate-power-manager #335: Display battery percentage

kikislater, 2020-03-17:
#### Expected behaviour
Display battery percentage near icon without clicking on it

#### Actual behaviour
No battery percentage available before clicking on icon

#### Steps to reproduce the behaviour
Why two applet for battery indicator ? One is complete and included in mate indicator applet but without direct view on battery percentage whereas other one is not included in mate indicator, display percentage and time remaining lacking battery stats and preferences ... 
[image]

#### MATE general version
1.24.0

#### Package version
1.24.0

#### Linux Distribution
Manjaro

#### Link to downstream report of your Distribution

andy9a9, 2020-06-09:
Maybe I didn't catch the right problem, but what's wrong here? 
[image]
Maybe I dind't

kikislater, 2020-06-10:
Interesting, I was thinking I had it in previous distribution and now I don't have it but I'm on Manjaro. May be they removed it. Any idea where to look ? mate-panel or mate-power-manager package ? 

[image]

Thank you

andy9a9, 2020-06-10:
I realized, that is handled by `/usr/lib/x86_64-linux-gnu/indicator-power/indicator-power-service` which is part of `indicator-power` package.
```
apt-cache policy indicator-power
indicator-power:
  Installed: 12.10.6+17.10.20170829.1-0ubuntu4
  Candidate: 12.10.6+17.10.20170829.1-0ubuntu4
  Version table:
 *** 12.10.6+17.10.20170829.1-0ubuntu4 500
```
So try to install this package.