mate-desktop/mate-power-manager #342: "Put display to sleep" timeout ignored

dark-penguin, 2020-04-26:
#### Expected behaviour
I set "Put display to sleep when inactive for:" to 30 minutes. Monitor turns off after 30 minutes.

#### Actual behaviour
I set "Put display to sleep when inactive for:" to anything. Monitor turns off after 10 minutes.

#### Steps to reproduce the behaviour
Set "Put display to sleep when inactive for:" to anything (I have not tried "Never" though). Monitor will turn off after 10 minutes.

#### MATE general version
1.20.4

#### Package version
1.20.3-2

#### Linux Distribution
Debian Buster

#### Link to downstream report of your Distribution
Does anyone care about bugs in Debian stable?..

This might be a problem with something else, but I have no idea how to debug it further.

I also set the "Screensaver" timeout ("Regard the computer as idle after:") to the same time, but disable both checkboxes ("Activate screensaver when computer is idle" and "Lock screen when screensaver is active").

lukefromdc, 2020-04-26:
I can confirm all of this except the time, which I've never measured.

On 4/26/2020 at 9:55 AM, "dark-penguin"  wrote:

	 Expected behaviour 

	I set "Put display to sleep when inactive for:" to 30 minutes.
Monitor turns off after 30 minutes. Actual behaviour 

	I set "Put display to sleep when inactive for:" to anything. Monitor
turns off after 10 minutes. Steps to reproduce the behaviour 

	Set "Put display to sleep when inactive for:" to anything (I have not
tried "Never" though). Monitor will turn off after 10 minutes. MATE
general version 

	1.20.4 Package version 

	1.20.3-2 Linux Distribution 

	Debian Buster Link to downstream report of your Distribution 

	Does anyone care about bugs in Debian stable?.. 

	This might be a problem with something else, but I have no idea how
to debug it further. 

	I also set the "Screensaver" timeout ("Regard the computer as idle
after:") to the same time, but disable both checkboxes ("Activate
screensaver when computer is idle" and "Lock screen when screensaver
is active").

voobscout, 2020-05-12:
Confirming this behavior with 1.24.1, built from https://pub.mate-desktop.org/releases/1.24/mate-power-manager-1.24.1.tar.xz

here's the dconf:
```ini
[org/mate/power-manager]
action-sleep-type-ac='nothing'
action-sleep-type-battery='nothing'
action-critical-battery='shutdown'
sleep-display-ac=0
sleep-display-battery=0
backlight-enable=false
brightness-ac=15.0
button-lid-ac='nothing'
button-lid-battery='nothing'
button-power='shutdown'
button-suspend='nothing'
idle-dim-time=0
idle-dim-ac=false
idle-dim-battery=false
dpms-method-ac='off'
dpms-method-battery='off'
notify-discharging=false
notify-fully-charged=false
notify-discharging=false
```

decrep, 2025-03-23:
I'm using mint 22.1 with mate 1.26.2, and I have this exact problem. 
Display goes to sleep after 10mins for all values over 10, even if "never" is set

lukefromdc, 2025-03-31:
For a desktop it should be enough to kill mate-power-manager and remove it from the list of startup applications. 

Do consider however that monitor's burning in seems to be back. I've seen this on both desktop and phone LCD screens so you really might not want to allow your desktop to show all day every day. Also keep anything privacy sensitive with a lot of contrast OFF the desktop as you really don't want it readable with the computer not even running

decrep, 2025-05-24:
I can't find how to kill mate power manager.
The trouble is the default dmps settings they are set to 600 seconds, which makes display turn off after 10mins.
But a temporary solution is to run this.
`xset dpms 18000 18000 18000

Which gives 3hrs on.
But it doesn't stick, has to be done every boot.
Any idea how to make it permanent?

decrep, 2025-05-26:
UPdate.
SMG on the linux mint forum solved it for me.
Just put
``xset dpms 18000 18000 18000`
inside you .profile file.