mate-desktop/mate-power-manager #350: Feature request: support for Wayland

Iey4iej3, 2020-07-29:
It seems to me that the current MATE power manager does not work under Wayland, even if XWayland is enabled. I tried that under `swaywm`.

lukefromdc, 2020-09-05:
I think I know why: the tray icon. Tray icons use GtkPlug/GtkSocket, which in turn relies on xembed, which is X11-only and does NOT work with xwayland 
Can you test again by first setting m-p-m to not show an icon, then trying to start it in your swaywm (or other wayland) session? If that works, supporting indicator mode for the icon could offer a way forward

rubyFeedback, 2021-12-07:
Perhaps that could be a meta-issue to see how much work it would be to have mate-desktop support wayland. Hopefully just as an option because I have been using xorg-server for so many years - I am a very lazy and selfish individual, I want others to smoothen out everything before I switch to anything else. :D

How does GNOME handle GtkSocket + wayland support for it?

mbkma, 2021-12-07:
We already have a tracker for wayland support, see https://wiki.mate-desktop.org/developers-corner/wayland-meson/
Some more information on that website would be nice of course...

wb9688, 2021-12-30:
@rubyFeedback: GNOME does not support tray, though there are shell extensions that add support for it.

Besides GtkSocket/GtkPlug, there is another way to implement tray: [Ayatana Indicators](https://ayatanaindicators.github.io/), which works using D-Bus.