mate-desktop/mate-power-manager #395: PowerManager-ERROR: Error in dbus when starting mate-power-preferences

thommignot1, 2024-02-21:
#### Expected behaviour
I expect mate-power-preferences to show up

#### Actual behaviour
```bash
$> mate-power-preferences 

(mate-power-preferences:2961): PowerManager-WARNING **: 19:01:31.129: Failed to get session for pid 2961: The name is not activatable

(mate-power-preferences:2961): PowerManager-ERROR **: 19:01:31.137: Error in dbus - GDBus.Error:org.freedesktop.DBus.Error.InvalidArgs: Invalid argument
Trace/breakpoint trap (core dumped)
```

#### Steps to reproduce the behaviour
Sorry I don't know how I came to this

#### MATE general version
```bash
$> pacman -Ss mate | grep installed
extra/caja 1.26.3-1 (mate) [installed]
extra/libmatekbd 1.26.0-1 [installed]
extra/libmatemixer 1.26.1-1 [installed]
extra/libmateweather 1.26.3-1 [installed]
extra/marco 1.26.2-2 (mate) [installed]
extra/mate-applets 1.26.1-1 (mate-extra) [installed]
extra/mate-backgrounds 1.26.0-1 (mate) [installed]
extra/mate-control-center 1.26.1-1 (mate) [installed]
extra/mate-desktop 1.26.2-1 (mate) [installed]
extra/mate-icon-theme 1.26.0-1 (mate) [installed]
extra/mate-media 1.26.2-1 (mate-extra) [installed]
extra/mate-menus 1.26.1-1 (mate) [installed]
extra/mate-notification-daemon 1.26.1-1 (mate) [installed]
extra/mate-panel 1.26.4-1 (mate) [installed]
extra/mate-polkit 1.26.1-1 (mate) [installed]
extra/mate-power-manager 1.26.1-1 (mate-extra) [installed]
extra/mate-screensaver 1.26.2-1 (mate-extra) [installed]
extra/mate-sensors-applet 1.26.0-2 (mate-extra) [installed]
extra/mate-session-manager 1.26.1-1 (mate) [installed]
extra/mate-settings-daemon 1.26.1-1 (mate) [installed]
extra/mate-themes 3.22.24-1 (mate) [installed]
extra/mate-user-guide 1.26.2-1 (mate) [installed]
```

#### Package version

#### Linux Distribution
```bash
$> uname -a
Linux warchy 6.7.5-arch1-1.1 # ...
```

#### Link to bugreport of your Distribution (requirement)
I don't know what that means ?

Hagar01, 2024-03-31:
Still in 1.28
dbus doesnt seem to be used when compiled for wayland.

sxl613, 2024-04-27:
Just as a +1, this happens for me too

```shell
$ mate-power-preferences

(mate-power-preferences:11877): PowerManager-WARNING **: 11:08:08.444: Failed to get session for pid 11877: The name is not activatable

(mate-power-preferences:11877): PowerManager-ERROR **: 11:08:08.459: Error in dbus - GDBus.Error:org.freedesktop.DBus.Error.FileNotFound: No such file or directory
zsh: trace trap (core dumped)  mate-power-preferences
```

```shell
$ uname -a
Linux sl 6.8.5-arch1-1 # ...
```

And general version information
```shell
$ pacman -Ss mate | grep installed
extra/atril 1.28.0-2 (mate-extra) [installed]
extra/caja 1.28.0-1 (mate) [installed]
extra/caja-image-converter 1.28.0-1 (mate-extra) [installed]
extra/caja-open-terminal 1.28.0-1 (mate-extra) [installed]
extra/caja-sendto 1.28.0-1 (mate-extra) [installed]
extra/caja-share 1.28.0-1 (mate-extra) [installed]
extra/caja-wallpaper 1.28.0-1 (mate-extra) [installed]
extra/caja-xattr-tags 1.28.0-1 (mate-extra) [installed]
extra/engrampa 1.28.1-1 (mate-extra) [installed]
extra/eom 1.28.0-1 (mate-extra) [installed]
extra/leiningen 2.10.0-2 [installed]
extra/libmatekbd 1.28.0-1 [installed]
extra/libmatemixer 1.28.0-1 [installed]
extra/libmateweather 1.28.0-1 [installed]
extra/marco 1.28.1-1 (mate) [installed]
extra/mate-applets 1.28.0-1 (mate-extra) [installed]
extra/mate-backgrounds 1.28.0-1 (mate) [installed]
extra/mate-calc 1.28.0-1 (mate-extra) [installed]
extra/mate-control-center 1.28.0-1 (mate) [installed]
extra/mate-desktop 1.28.2-1 (mate) [installed]
extra/mate-icon-theme 1.28.0-1 (mate) [installed]
extra/mate-icon-theme-faenza 1.20.0-2 (mate-extra) [installed]
extra/mate-media 1.28.1-1 (mate-extra) [installed]
extra/mate-menus 1.28.0-1 (mate) [installed]
extra/mate-netbook 1.26.0-1 (mate-extra) [installed]
extra/mate-notification-daemon 1.28.0-1 (mate) [installed]
extra/mate-panel 1.28.0-2 (mate) [installed]
extra/mate-polkit 1.28.1-1 (mate) [installed]
extra/mate-power-manager 1.28.1-1 (mate-extra) [installed]
extra/mate-screensaver 1.28.0-1 (mate-extra) [installed]
extra/mate-sensors-applet 1.28.0-1 (mate-extra) [installed]
extra/mate-session-manager 1.28.0-1 (mate) [installed]
extra/mate-settings-daemon 1.28.0-1 (mate) [installed]
extra/mate-system-monitor 1.28.1-1 (mate-extra) [installed]
extra/mate-terminal 1.28.1-1 (mate-extra) [installed]
extra/mate-themes 3.22.24-1 (mate) [installed]
extra/mate-user-guide 1.28.0-1 (mate) [installed]
extra/mate-user-share 1.28.0-1 (mate-extra) [installed]
extra/mate-utils 1.28.0-1 (mate-extra) [installed]
extra/mozo 1.28.0-1 (mate-extra) [installed]
extra/perl-www-mechanize 2.18-1 [installed]
extra/pluma 1.28.0-2 (mate-extra) [installed]
```

centic9-dt, 2026-11-02:
Had the same crash caused by an invalid `resume=` in the kernel commandline and also think it should be improved in mate-power-manager instead of silently crashing when invoked via the control center.

Steps to fix it for me on Ubuntu Noble (24.04): 
* Remove `resume=` and `resume_offset=` from `/etc/default/grub` 
* Run `sudo grub-install`
* Run `sudo update-grub`
* reboot the computer
* => only then preferences worked again

For the record, the error message dbus-send returns is `Failed to create bus message: Invalid argument` which is also very unhelpful.