mate-desktop/mate-power-manager #398: Inhibit Applet doesn't work with Crossover ...

Omnimaxus, 2024-04-09:
#### Expected behaviour

Keep the monitor on and inhibit power management when running an application via CodeWeavers CrossOver.

#### Actual behaviour

Does not keep the monitor on.  Monitor goes off.  No inhibition of power management is happening.

#### Steps to reproduce the behaviour

Start Crossover.  Run an application.  Wait and observe.

#### MATE general version

N/A

#### Package version

Latest one (?) as of 4/9/24.

#### Linux Distribution

Linux Mint 21.3 Cinnamon (Edge)

#### Link to bugreport of your Distribution (requirement)

Not sure how to do this, but ... the described behavior is true and should be fixed.  (Thank you.)

lukefromdc, 2024-04-14:
I have no way of testing this directly as I do not use paid/closed software. If anyone else on the team has or can test Crossover (or if someone at Crossover can work on this and post a PR) they might be able to test it. Also, if this happens with an older version of Crossover someone running WINE (which as far as I know gets Crossover's code with a delay) might be able to duplicate it.

HOWEVER, the inhibit applet should always be able to prevent the monitor from turning off without regard for what if any toplevel software is running. Does this occur only when Crossover is in use, or does this also happen with your system idling at the desktop?

For something like Crossover to cause the machine to stay awake would not be a surprise, but for it to prevent the OS from keeping the machine awake would be quite odd.

Omnimaxus, 2024-06-20:
I don't know.  But thanks for replying.