mate-desktop/mate-power-manager #402: Battery symbol disappears after a while

munzelthecat, 2024-10-05:
#### Expected behaviour

#### Actual behaviour
On my Notebook, after a while the battery symbol in the bar disappears. 

#### Steps to reproduce the behaviour
Work with the notebook and wait.

#### MATE general version

Mate 1.26.2

#### Package version
mate-power-manager 1.26.1+mint1

#### Linux Distribution
Linux Mint 22 Mate 

#### Link to bugreport of your Distribution (requirement)
https://termbin.com/xobh

lukefromdc, 2024-10-12:
If you open mate-system-monitor and go to the "processes" tab, you can see if mate-system-monitor is still running or not. If mte-power-manager does not show up in this list of processes, it has crashed, or maybe frozen. 

Try running `killall -9 mate-power-manager ` first, then `mate-power-manager` and if this does not bring back the battery icon, please post whatever text you get in the terminal window from the second command

munzelthecat, 2024-10-14:
Hello,

here are the messages after `mate-power-manager`

```
(mate-power-manager:3469): PowerManager-WARNING **: 18:23:52.231: Failed to get session for pid 3469: The name org.freedesktop.ConsoleKit was not provided by any .service files

(mate-power-manager:3469): PowerManager-WARNING **: 18:23:52.236: could not map keysym 1008ffa8 to keycode
```

lukefromdc, 2024-10-14:
Neither of those should cause a delayed crash. Does it keep running in terminal,
or does it exit?

munzelthecat, 2024-10-15:
As long as the terminal remains open, the battery symbol will be visible. If the terminal is closed, it disappears.

munzelthecat, 2024-10-15:
It seems to be a problem where the hardware could play a role, or there are two problems.

I noticed today that the battery symbol did not disappear, but the charge level did not change for a long time and the battery LED on the notebook began to flash (which remains so even after a shutdown). Also `acpi -V` showed the wrong charge level.  
With
`killall -9 mate-power-manager` I ended the battery display and then ran 
`mate-power-manager.` The error message then also appeared.

```
~$ mate-power-manager

(mate-power-manager:5036): PowerManager-WARNING **: 20:20:53.960: Failed to get session for pid 5036: The name org.freedesktop.ConsoleKit was not provided by any .service files

(mate-power-manager:5036): PowerManager-WARNING **: 20:20:53.965: could not map keysym 1008ffa8 to keycode
```

lukefromdc, 2024-10-15:
If this is being caused by a hardware issue, that could mean nobody without the same hardware problem can duplicate it. Seems to me there is almost always something on a laptop whose makers only support Windows that doesn't work right or doesn't work at all in Linux. I've had two HP laptops in a row where the touchpad isn't detected at all so a separate mouse is required.

munzelthecat, 2024-10-19:
Maybe it is a hardware issue. With `dmesg` I could find this:
```
[ 1335.708554] mate-power-mana[2014]: segfault at 7265706c7a ip 0000785dcb3cacd8 sp 00007ffd331ceb88 error 4 in libgobject-2.0.so.0.8000.0[785dcb39d000+37000] likely on CPU 1 (core 0, socket 0)
[ 1335.708573] Code: 8d 15 8c 54 02 00 48 c1 e9 02 4c 8b 0c ca 4d 85 c9 74 07 41 f6 41 16 20 75 36 48 8b 38 48 81 ff fc 03 00 00 76 3c 48 83 e7 fc <f6> 47 16 10 74 05 4d 85 c9 75 4d 31 c0 c3 66 2e 0f 1f 84 00 00 00
```

munzelthecat, 2024-10-25:
I would like to add something else. I now have the third battery in my notebook and it seems to be working smoothly with this one. Thank you very much for your thoughts.

Best regards
Munzelthecat

bet5y, 2024-10-28:
This has just happened to me - one minute the battery symbol was there, the next it had gone. Hopefully the following will be of assistance.

https://termbin.com/euey

bet5y, 2024-10-31:
Another crash:

https://termbin.com/3lr6

I ran the following as mentioned above after the above crash:

`colin@ASUS-LM22-Mate:~$ mate-power-manager &
[1] 8264
colin@ASUS-LM22-Mate:~$ 
(mate-power-manager:8264): PowerManager-WARNING **: 00:21:31.395: Failed to get session for pid 8264: The name org.freedesktop.ConsoleKit was not provided by any .service files

(mate-power-manager:8264): PowerManager-WARNING **: 00:21:31.401: could not map keysym 1008ffa8 to keycode
`
I also ran the following:

`colin@ASUS-LM22-Mate:~$ killall -9 mate-power-manager
colin@ASUS-LM22-Mate:~$ mate-power-manager

(mate-power-manager:8406): PowerManager-WARNING **: 00:23:26.162: Failed to get session for pid 8406: The name org.freedesktop.ConsoleKit was not provided by any .service files

(mate-power-manager:8406): PowerManager-WARNING **: 00:23:26.169: could not map keysym 1008ffa8 to keycode
`

soadroxs, 2025-01-14:
Hi! I'm also having this issue with a Lenovo Thinkpad T430 and a SANYO 45N1001 battery. Here are the logs from the crash.

https://termbin.com/xvig

Mupli, 2025-02-02:
I have the same issue on x13 / ryzen 7 pro 4750u. 
mate-power-manager: **1.26.1+mint1**

https://termbin.com/tj3b

sk8board, 2025-06-06:
Same issue. I have three of these computers but the problem is only present on one computer.

> System:
  Kernel: 6.8.0-51-generic arch: x86_64 bits: 64 compiler: gcc v: 13.3.0 clocksource: tsc
  Desktop: MATE v: 1.26.2 wm: marco v: 1.26.2 with: mate-panel tools: mate-screensaver
    dm: LightDM v: 1.30.0 Distro: Linux Mint 22.1 Xia base: Ubuntu 24.04 noble
Machine:
  Type: Laptop System: Dell product: Latitude 7390 v: N/A serial: <superuser required> Chassis:
    type: 10 serial: <superuser required>
  Mobo: Dell model: N/A serial: <superuser required> part-nu: 081B uuid: <superuser required>
    UEFI: Dell v: 1.43.0 date: 03/06/2025
Battery:
  ID-1: BAT0 charge: 36.3 Wh (59.7%) condition: 60.8/60.8 Wh (100.0%) volts: 7.6 min: 7.6
    model: LGC-LGC7.780 DELL 2X39G type: Li-ion serial: <filter> status: discharging
CPU:
  Info: quad core model: Intel Core i7-8650U bits: 64 type: MT MCP smt: enabled arch: Coffee Lake
    rev: A cache: L1: 256 KiB L2: 1024 KiB L3: 8 MiB
  Speed (MHz): avg: 1450 high: 1600 min/max: 400/4200 cores: 1: 1600 2: 400 3: 1600 4: 1600
    5: 1600 6: 1600 7: 1600 8: 1600 bogomips: 33599
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx