mate-desktop/mate-power-manager #406: Lot of enum unrecognized 17 in xsession-errors file

coozoo, 2025-01-24:
Recently I have find out that my .xsession-errors file some kinda large

I've checked and most of lines related to power manager

```
(mate-power-manager:1108910): PowerManager-WARNING **: 00:21:45.210: enum unrecognised: 17
(mate-power-manager:1108910): PowerManager-WARNING **: 00:21:45.211: in an undefined state we are not charging or discharging and the batteries are also not charged
(mate-power-manager:1108910): PowerManager-WARNING **: 00:21:45.211: enum unrecognised: 17
(mate-power-manager:1108910): PowerManager-WARNING **: 00:21:45.211: in an undefined state we are not charging or discharging and the batteries are also not charged
(mate-power-preferences:3779243): PowerManager-WARNING **: 00:22:06.852: Failed to get session for pid 3779243: The name is not activatable
(mate-power-manager:1108910): PowerManager-WARNING **: 00:22:15.282: enum unrecognised: 17
(mate-power-manager:1108910): PowerManager-WARNING **: 00:22:15.282: in an undefined state we are not charging or discharging and the batteries are also not charged
(mate-power-manager:1108910): PowerManager-WARNING **: 00:22:15.283: enum unrecognised: 17
(mate-power-manager:1108910): PowerManager-WARNING **: 00:22:15.283: in an undefined state we are not charging or discharging and the batteries are also not charged
(mate-power-statistics:3779287): PowerManager-WARNING **: 00:22:21.771: nothing matched, falling back to default icon

```

dragonfly7-lnx, 2025-12-18:
@lukefromdc Thanks for getting back to me ! I appreciate it very much !

Actually your statements sound reasonable. The hardware is a Fujitsu Lifebook U749 (~ 5...6 yrs old). It looks like that notebook has the latest firmware. So I doubt that there will be a fixed firmware waiting for me to solve my current issue somewhere.

Just a sidenote: Any indication along power management looks ok, i.e. battery SOC, indications of charging / discharging and projected remaining "battery time" are ok. So no obvious indication on this side - at least not in my understanding.

I'll do a deep dive in the UEFI settings although I did not do any special setting. Actually I did buy this refurbished notebook a year ago in this win11 compatibility hype. It has got an i5-8xxx which was mentioned to be the oldest officially supported intel CPU. The very 1st step after receiving this machine was resetting UEFI values to their default values. I did not alter any pwrmgmt settings back then.

One tiny sidenote: I'm running cinammon 6.4 as part of LMDE7. I'm not using MATE. But the indications seem very similar.

lukefromdc, 2025-12-19:
I am no expert in laptop power management in general, this sounds like battery percentage itself reads OK but some piece of the
"state" data is getting corrupted.  If you are able to read battery charge level and whether charging or not, I would advise the 
workaround of a script that sleeps most of the time but deletes the growing .xsession-errors file at intervals. If you need to debug
another session error just kill the script and wait for the problem being diagnosed to show up.

SamSirry, 2026-09-08:
FWIW, I found out these logged entries are only showing while a bluetooth headset is connected.
Apparently, this is caused by the bluetooth manager reading the headset's battery level every ~15 seconds.

SamSirry, 2026-09-08:
Relevant: https://github.com/linuxmint/Cinnamon/issues/13433

lukefromdc, 2026-09-13:
We do not provide the bluetooth manager (blueman) applet, GNOME does. The battery reading intervals are reasonable, but it should write to the logs once only that it cannot read the data when this is the case. This is almost certainly a firmware bug.

I do not know how that's reaching mate-power-manager though and I use only wired headphones myself.